Nitrokey/nitrokey-storage-firmware #1: Device lockup during storage initialization

msgctl, 2015-09-15:
In the middle of random prefill the device turned unresponsive. Right
after reconnecting the device, Nitrokey application informed me of
underlying storage being fully prefilled, which wasn't true.

Excerpt from dmesg:
[179124.436222] usb 1-1.2: reset high-speed USB device number 102 using ehci-pci
[179124.724359] usb 1-1.2: device descriptor read/64, error -71
[179139.902709] usb 1-1.2: device descriptor read/64, error -110
[179140.078757] usb 1-1.2: reset high-speed USB device number 102 using ehci-pci
[179155.157304] usb 1-1.2: device descriptor read/64, error -110
[179170.335703] usb 1-1.2: device descriptor read/64, error -110
[179170.511653] usb 1-1.2: reset high-speed USB device number 102 using ehci-pci
[179180.921144] usb 1-1.2: device not accepting address 102, error -110
[179180.993440] usb 1-1.2: reset high-speed USB device number 102 using ehci-pci
[179191.402844] usb 1-1.2: device not accepting address 102, error -110
[179191.403670] usb 1-1.2: USB disconnect, device number 102
[179191.483159] usb 1-1.2: new high-speed USB device number 103 using ehci-pci
[179206.557575] usb 1-1.2: device descriptor read/64, error -110
[179221.735995] usb 1-1.2: device descriptor read/64, error -110
[179221.912112] usb 1-1.2: new high-speed USB device number 104 using ehci-pci

msgctl, 2015-09-15:
Nitrokey Storage reconnects frequently during random prefill, but the application appears to continue nonetheless.

Test environment information:
00:1d.0 USB controller: Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controller #1 (rev 04)

Distributor ID:	Debian
Description:	Debian GNU/Linux 6.0.10 (squeeze)
Release:	6.0.10
Codename:	squeeze

Excerpt from dmesg:
[178864.786062] usb 1-1.2: reset high-speed USB device number 102 using ehci-pci
[178864.883062] input: NitroKey NitroKey Storage as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.2/0003:20A0:4109.010D/input/input287
[178864.883215] hid-generic 0003:20A0:4109.010D: input,hidraw0: USB HID v1.10 Keyboard [NitroKey NitroKey Storage] on usb-0000:00:1a.0-1.2/input2
[178864.884103] usb 1-1.2: usbfs: process 28944 (CryptoStickUtil) did not claim interface 2 before use
[178864.966072] usb 1-1.2: reset high-speed USB device number 102 using ehci-pci
[178865.061043] usb 1-1.2: usbfs: process 28944 (CryptoStickUtil) did not claim interface 2 before use
[178865.061101] usb 1-1.2: usbfs: process 28944 (CryptoStickUtil) did not claim interface 2 before use
[178865.142118] usb 1-1.2: reset high-speed USB device number 102 using ehci-pci
[178865.236998] usb 1-1.2: usbfs: process 28944 (CryptoStickUtil) did not claim interface 2 before use
[178865.237076] usb 1-1.2: usbfs: process 28944 (CryptoStickUtil) did not claim interface 2 before use
[178865.237081] usb 1-1.2: usbfs: process 28944 (CryptoStickUtil) did not claim interface 2 before use
[178865.374260] usb 1-1.2: reset high-speed USB device number 102 using ehci-pci
[178865.470990] input: NitroKey NitroKey Storage as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.2/0003:20A0:4109.010E/input/input288
[178865.471165] hid-generic 0003:20A0:4109.010E: input,hidraw0: USB HID v1.10 Keyboard [NitroKey NitroKey Storage] on usb-0000:00:1a.0-1.2/input2
[178865.474041] usb 1-1.2: usbfs: process 28944 (CryptoStickUtil) did not claim interface 2 before use
[178865.477758] usb 1-1.2: usbfs: process 28944 (CryptoStickUtil) did not claim interface 2 before use
[178865.554224] usb 1-1.2: reset high-speed USB device number 102 using ehci-pci
[178865.649002] usb 1-1.2: usbfs: process 28944 (CryptoStickUtil) did not claim interface 2 before use
[178865.730234] usb 1-1.2: reset high-speed USB device number 102 using ehci-pci
[178865.825530] usb 1-1.2: usbfs: process 28944 (CryptoStickUtil) did not claim interface 2 before use
[178865.825586] usb 1-1.2: usbfs: process 28944 (CryptoStickUtil) did not claim interface 2 before use
[178865.897724] usb 1-1.2: usbfs: process 28944 (CryptoStickUtil) did not claim interface 2 before use
[178865.998911] usb 1-1.2: usbfs: process 28944 (CryptoStickUtil) did not claim interface 2 before use
[178866.099097] usb 1-1.2: usbfs: process 28944 (CryptoStickUtil) did not claim interface 2 before use
[178866.198294] usb 1-1.2: usbfs: process 28944 (CryptoStickUtil) did not claim interface 2 before use
[178866.198307] usb 1-1.2: usbfs: process 28944 (CryptoStickUtil) did not claim interface 2 before use
[178866.342445] usb 1-1.2: reset high-speed USB device number 102 using ehci-pci
[178866.438666] input: NitroKey NitroKey Storage as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.2/0003:20A0:4109.010F/input/input289
[178866.438812] hid-generic 0003:20A0:4109.010F: input,hidraw0: USB HID v1.10 Keyboard [NitroKey NitroKey Storage] on usb-0000:00:1a.0-1.2/input2
[178866.441690] usb 1-1.2: usbfs: process 28944 (CryptoStickUtil) did not claim interface 2 before use
[178866.445514] usb 1-1.2: usbfs: process 28944 (CryptoStickUtil) did not claim interface 2 before use
[178866.522362] usb 1-1.2: reset high-speed USB device number 102 using ehci-pci
[178866.617047] usb 1-1.2: usbfs: process 28944 (CryptoStickUtil) did not claim interface 2 before use

jans23, 2015-09-15:
Thanks for the report. Does the same device behave similar at another computer (hardware)?

msgctl, 2015-09-16:
@jans23 Yes, it's the same with:
00:04.0 USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #1 (rev 03)
00:04.1 USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #2 (rev 03)
00:04.2 USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #3 (rev 03)
00:04.7 USB controller: Intel Corporation 82801I (ICH9 Family) USB2 EHCI Controller #1 (rev 03)

(USB 2.0)

szszszsz, 2016-06-07:
Possibly fixed in v0.34. Will be reopened if occur again.

alem0lars, 2016-11-03:
After initialization I'm still getting the errors:

```
[12032.636569] usb 3-2: usbfs: process 6384 (nitrokey-app) did not claim interface 2 before use
[12034.636204] usb 3-2: usbfs: process 6384 (nitrokey-app) did not claim interface 2 before use
[12042.994328] usb 3-2: reset high-speed USB device number 20 using xhci_hcd
[12043.163194] input: Nitrokey Nitrokey Storage as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.2/0003:20A0:4109.0031/input/input55
[12043.214398] hid-generic 0003:20A0:4109.0031: input,hidraw1: USB HID v1.10 Keyboard [Nitrokey Nitrokey Storage] on usb-0000:00:14.0-2/input2
[12044.635289] usb 3-2: usbfs: process 6384 (nitrokey-app) did not claim interface 2 before use
[12074.016118] usb 3-2: reset high-speed USB device number 20 using xhci_hcd
[12074.184675] input: Nitrokey Nitrokey Storage as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.2/0003:20A0:4109.0032/input/input56
[12074.235094] hid-generic 0003:20A0:4109.0032: input,hidraw1: USB HID v1.10 Keyboard [Nitrokey Nitrokey Storage] on usb-0000:00:14.0-2/input2
[12074.633427] usb 3-2: usbfs: process 11033 (nitrokey-app) did not claim interface 2 before use
[12076.635627] usb 3-2: usbfs: process 11033 (nitrokey-app) did not claim interface 2 before use
[12104.989654] usb 3-2: reset high-speed USB device number 20 using xhci_hcd
[12105.158167] input: Nitrokey Nitrokey Storage as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.2/0003:20A0:4109.0033/input/input57
[12105.208640] hid-generic 0003:20A0:4109.0033: input,hidraw1: USB HID v1.10 Keyboard [Nitrokey Nitrokey Storage] on usb-0000:00:14.0-2/input2
[12106.631207] usb 3-2: usbfs: process 11033 (nitrokey-app) did not claim interface 2 before use
[12108.630576] usb 3-2: usbfs: process 11033 (nitrokey-app) did not claim interface 2 before use
[12136.027347] usb 3-2: reset high-speed USB device number 20 using xhci_hcd
[12136.196184] input: Nitrokey Nitrokey Storage as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.2/0003:20A0:4109.0034/input/input58
[12136.247307] hid-generic 0003:20A0:4109.0034: input,hidraw1: USB HID v1.10 Keyboard [Nitrokey Nitrokey Storage] on usb-0000:00:14.0-2/input2
[12136.628748] usb 3-2: usbfs: process 11033 (nitrokey-app) did not claim interface 2 before use
```

making the nitrokey storage unusable.

Firmware version is 0.43 (latest).

I'm using Gentoo Linux.

szszszsz, 2016-11-03:
Hi! Thank you for reporting! 
Could you share the kernel version?

alem0lars, 2016-11-03:
I'm using kernel `4.4.26`:

```
$ uname -a
Linux kate 4.4.26-gentoo #6 SMP Thu Nov 3 21:02:48 CET 2016 x86_64 Intel(R) Core(TM) i7-4710MQ CPU @ 2.50GHz GenuineIntel GNU/Linux
```

alem0lars, 2016-11-03:
After a lot of connect/disconnect cycles (due to the errors posted before) I successfully initialized the key (took about 1h).

However I'm still getting these errors (always).

jans23, 2016-11-03:
Do you use a USB 2.0 or USB 3 port? If possible, try a port with a different USB version.

alem0lars, 2016-11-03:
I'm getting the same error with both usb2 and usb3.

It seems the usb 3 driver is used `xhci_hcd` (see line `1719.532743`):

```
[ 1702.492236] usb 3-9.4.1: New USB device found, idVendor=20a0, idProduct=4109
[ 1702.492240] usb 3-9.4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1702.492242] usb 3-9.4.1: Product: Nitrokey Storage
[ 1702.492243] usb 3-9.4.1: Manufacturer: Nitrokey
[ 1702.492245] usb 3-9.4.1: SerialNumber: 0000000000000
[ 1702.493635] usb-storage 3-9.4.1:1.0: USB Mass Storage device detected
[ 1702.493719] scsi host10: usb-storage 3-9.4.1:1.0
[ 1702.496375] input: Nitrokey Nitrokey Storage as /devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.4/3-9.4.1/3-9.4.1:1.2/0003:20A0:4109.0016/input/input29
[ 1702.547406] hid-generic 0003:20A0:4109.0016: input,hidraw1: USB HID v1.10 Keyboard [Nitrokey Nitrokey Storage] on usb-0000:00:14.0-9.4.1/input2
[ 1702.668438] e1000e: enp0s25 NIC Link is Up 100 Mbps Full Duplex, Flow Control: Rx/Tx
[ 1702.668444] e1000e 0000:00:19.0 enp0s25: 10/100 speed: disabling TSO
[ 1703.497296] scsi 10:0:0:0: Direct-Access     Nitrokey Nitrokey Storage 1.00 PQ: 0 ANSI: 0
[ 1703.497461] sd 10:0:0:0: Attached scsi generic sg2 type 0
[ 1703.501672] scsi 10:0:0:1: Direct-Access     Nitrokey Nitrokey Storage 1.00 PQ: 0 ANSI: 0
[ 1703.501783] sd 10:0:0:1: Attached scsi generic sg3 type 0
[ 1703.501802] sd 10:0:0:0: [sdc] 4194304 512-byte logical blocks: (2.15 GB/2.00 GiB)
[ 1703.504734] sd 10:0:0:0: [sdc] Write Protect is off
[ 1703.504740] sd 10:0:0:0: [sdc] Mode Sense: 1b 00 00 00
[ 1703.507725] sd 10:0:0:0: [sdc] No Caching mode page found
[ 1703.507728] sd 10:0:0:0: [sdc] Assuming drive cache: write through
[ 1703.523266] sd 10:0:0:1: [sdd] Attached SCSI removable disk
[ 1703.528002]  sdc: sdc1
[ 1719.532743] usb 3-9.4.1: reset high-speed USB device number 15 using xhci_hcd
[ 1719.621590] input: Nitrokey Nitrokey Storage as /devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.4/3-9.4.1/3-9.4.1:1.2/0003:20A0:4109.0017/input/input30
[ 1719.672799] hid-generic 0003:20A0:4109.0017: input,hidraw1: USB HID v1.10 Keyboard [Nitrokey Nitrokey Storage] on usb-0000:00:14.0-9.4.1/input2
[ 1720.238666] usb 3-9.4.1: usbfs: process 6511 (nitrokey-app) did not claim interface 2 before use
[ 1722.239776] usb 3-9.4.1: usbfs: process 6511 (nitrokey-app) did not claim interface 2 before use
```

jans23, 2016-11-04:
Do you use the latest version of NK App? If not, please try again using it.

alem0lars, 2016-11-04:
Yes, I'm using the latest version (`0.5.1`).

alem0lars, 2016-11-11:
Are there any news on this issue?

I'm unable to use the Nitrokey Storage

jans23, 2016-11-11:
No news yet. Please give us more time.

alem0lars, 2016-11-11:
Ok no problem :)

Please let me know if I can help, providing some specific information and/or test possible fixes.

Thanks

szszszsz, 2016-12-01:
Hi! Sorry for long delay.
Looking at the latest log device is being reset by the OS just (16 secs) after insertion:
```
[ 1703.528002]  sdc: sdc1
[ 1719.532743] usb 3-9.4.1: reset high-speed USB device number 15 using xhci_hcd
```
so later references to it from the App are reported as invalid in dmesg log. I would suspect some system service trying to access the device and resetting it when it is owned already (by App). 
Maybe scdaemon? Could you check working of the App in a somewhat limited environment?

alem0lars, 2016-12-04:
I can't get it working. 

Are there any particular kernel options that need to be enabled?

szszszsz, 2016-12-05:
Thank you for checking. Could you write briefly what have you checked lately?
I think next step would be preparing some CLI tool for communication test. I will let you know once its ready.

alem0lars, 2016-12-31:
Are there any news? I'm still unable to use it due to continuous device disconnections.. 

What are the kernel modules needed to fully support nitrokey storage? (I'm a gentoo user so I need to manually select the needed kernel modules)

szszszsz, 2017-01-02:
The debug log you have pasted suggest some access conflict between the applications or issuing `reset` command by the OS for unknown reason. 
Could you reinsert the device in single user mode and attach the log (as a file should be most handy) from next 60 seconds (`init 1`)?
Please repeat this step in GUI disabled (`init 3`).

As for kernel modules, I think you have all needed - usb_storage and hid are mentioned in your log. Here is my modules list with inserted Storage: http://paste.ubuntu.com/23727275/

szszszsz, 2017-08-19:
Will be tested on next development version of 0.47.

szszszsz, 2017-08-23:
Hi @alem0lars !
Could you see how the things are on v0.48? The USB communication has been improved lately.

szszszsz, 2017-10-25:
It would be great to retest this issue on latest software. Could you take it @techge ?

techge, 2017-10-30:
Mh, I am not sure, to be honest. This conversation sounds like the bug is difficult to reproduce? I will try...

So firmware <0.47 and possibly kernel 4.4?

alem0lars, 2017-10-30:
I'm going to give another try to the nitrokey in the afternoon.

I'll update you with the state of the issue (if it's still present or
not) in a couple of days.

alem0lars, 2017-10-30:
During storage initialization I get the following errors:

```
Oct 30 13:22:36 kate kernel: usb 3-3: reset high-speed USB device number 25 using xhci_hcd
Oct 30 13:22:36 kate kernel: input: Nitrokey Nitrokey Storage as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.2/0003:20A0:4109.013
Oct 30 13:22:36 kate kernel: hid-generic 0003:20A0:4109.013C: input,hidraw2: USB HID v1.10 Keyboard [Nitrokey Nitrokey Storage] on us
Oct 30 13:22:38 kate kernel: usb 3-3: usbfs: process 13883 (QThread) did not claim interface 2 before use
Oct 30 13:23:06 kate kernel: usb 3-3: reset high-speed USB device number 25 using xhci_hcd
Oct 30 13:23:07 kate kernel: input: Nitrokey Nitrokey Storage as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.2/0003:20A0:4109.013
Oct 30 13:23:07 kate kernel: hid-generic 0003:20A0:4109.013D: input,hidraw2: USB HID v1.10 Keyboard [Nitrokey Nitrokey Storage] on us
Oct 30 13:23:27 kate kernel: usb 3-3: usbfs: process 15530 (QThread) did not claim interface 2 before use
Oct 30 13:23:38 kate kernel: usb 3-3: reset high-speed USB device number 25 using xhci_hcd
Oct 30 13:23:38 kate kernel: input: Nitrokey Nitrokey Storage as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.2/0003:20A0:4109.013
Oct 30 13:23:38 kate kernel: hid-generic 0003:20A0:4109.013E: input,hidraw2: USB HID v1.10 Keyboard [Nitrokey Nitrokey Storage] on us
Oct 30 13:23:46 kate kernel: usb 3-3: usbfs: process 15664 (QThread) did not claim interface 2 before use
Oct 30 13:24:08 kate kernel: usb 3-3: reset high-speed USB device number 25 using xhci_hcd
Oct 30 13:24:09 kate kernel: input: Nitrokey Nitrokey Storage as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.2/0003:20A0:4109.013
Oct 30 13:24:09 kate kernel: hid-generic 0003:20A0:4109.013F: input,hidraw2: USB HID v1.10 Keyboard [Nitrokey Nitrokey Storage] on us
Oct 30 13:24:13 kate kernel: usb 3-3: usbfs: process 15931 (QThread) did not claim interface 2 before use
```

However I managed to complete the initialization.

---

Now I'm getting the following error when I plug nitrokey storage with application already open:

```
Oct 30 13:24:39 kate kernel: usb-storage 3-3:1.0: USB Mass Storage device detected
Oct 30 13:24:39 kate kernel: scsi host6: usb-storage 3-3:1.0
Oct 30 13:24:39 kate kernel: input: Nitrokey Nitrokey Storage as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.2/0003:20A0:4109.014
Oct 30 13:24:39 kate kernel: hid-generic 0003:20A0:4109.0140: input,hidraw2: USB HID v1.10 Keyboard [Nitrokey Nitrokey Storage] on us
Oct 30 13:24:39 kate systemd[1]: Reached target Smart Card.
Oct 30 13:24:40 kate kernel: scsi 6:0:0:0: Direct-Access     Nitrokey Nitrokey Storage 1.00 PQ: 0 ANSI: 0
Oct 30 13:24:40 kate kernel: sd 6:0:0:0: Attached scsi generic sg2 type 0
Oct 30 13:24:40 kate kernel: scsi 6:0:0:1: Direct-Access     Nitrokey Nitrokey Storage 1.00 PQ: 0 ANSI: 0
Oct 30 13:24:40 kate kernel: sd 6:0:0:0: [sdc] 4194304 512-byte logical blocks: (2.15 GB/2.00 GiB)
Oct 30 13:24:40 kate kernel: sd 6:0:0:1: Attached scsi generic sg3 type 0
Oct 30 13:24:40 kate kernel: sd 6:0:0:0: [sdc] Write Protect is off
Oct 30 13:24:40 kate kernel: sd 6:0:0:0: [sdc] Mode Sense: 1b 00 00 00
Oct 30 13:24:40 kate kernel: sd 6:0:0:0: [sdc] No Caching mode page found
Oct 30 13:24:40 kate kernel: sd 6:0:0:0: [sdc] Assuming drive cache: write through
Oct 30 13:24:40 kate kernel: sd 6:0:0:1: [sdd] Attached SCSI removable disk
Oct 30 13:24:40 kate kernel:  sdc: sdc1
Oct 30 13:24:56 kate kernel: usb 3-3: reset high-speed USB device number 26 using xhci_hcd
Oct 30 13:25:02 kate kernel: usb 3-3: device descriptor read/64, error -110
Oct 30 13:25:18 kate kernel: usb 3-3: device descriptor read/64, error -110
Oct 30 13:25:18 kate kernel: usb 3-3: reset high-speed USB device number 26 using xhci_hcd
Oct 30 13:25:24 kate kernel: usb 3-3: device descriptor read/64, error -110
Oct 30 13:25:39 kate kernel: usb 3-3: device descriptor read/64, error -110
Oct 30 13:25:40 kate kernel: usb 3-3: reset high-speed USB device number 26 using xhci_hcd
Oct 30 13:25:41 kate systemd-udevd[497]: seq 11949 '/devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/host6/target6:0:0/6:0:0:1/block
Oct 30 13:25:41 kate systemd-udevd[497]: seq 11952 '/devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/host6/target6:0:0/6:0:0:0/block
Oct 30 13:25:45 kate kernel: xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
Oct 30 13:25:51 kate kernel: xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
Oct 30 13:25:51 kate kernel: usb 3-3: device not accepting address 26, error -62
Oct 30 13:25:51 kate kernel: usb 3-3: reset high-speed USB device number 26 using xhci_hcd
Oct 30 13:25:56 kate kernel: xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
Oct 30 13:26:02 kate kernel: xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
Oct 30 13:26:02 kate systemd-udevd[16421]: Process '/sbin/hdparm -B 254 -S 192 /dev/sdd' failed with exit code 22.
Oct 30 13:26:02 kate kernel: usb 3-3: device not accepting address 26, error -62
Oct 30 13:26:02 kate kernel: usb 3-3: USB disconnect, device number 26
Oct 30 13:26:02 kate kernel: sd 6:0:0:1: [sdd] tag#0 FAILED Result: hostbyte=DID_NO_CONNECT driverbyte=DRIVER_OK
Oct 30 13:26:02 kate kernel: sd 6:0:0:1: [sdd] tag#0 CDB: ATA command pass through(16) 85 06 20 00 00 00 c0 00 00 00 00 00 00 40 e3 0
Oct 30 13:26:02 kate kernel: sd 6:0:0:1: [sdd] tag#0 FAILED Result: hostbyte=DID_NO_CONNECT driverbyte=DRIVER_OK
[... cut ...]
Oct 30 13:26:02 kate systemd-udevd[17605]: inotify_add_watch(9, /dev/sdc1, 10) failed: No such file or directory
Oct 30 13:26:02 kate systemd-udevd[17605]: Process '/sbin/hdparm -B 254 -S 192 /dev/sdc' failed with exit code 6.
Oct 30 13:26:02 kate systemd[1]: smartcard.target: Unit not needed anymore. Stopping.
Oct 30 13:26:02 kate systemd[1]: Stopped target Smart Card.
Oct 30 13:26:02 kate kernel: usb 3-3: new high-speed USB device number 27 using xhci_hcd
Oct 30 13:26:08 kate kernel: usb 3-3: device descriptor read/64, error -110
```

szszszsz, 2017-10-30:
Hi! Thank you for retesting. Could you update your environment details for completeness? That is:
- Storage firmware version
- Nitrokey App version
- OS and kernel version 

Would it be possible to compile the Nitrokey App code from Git repository, `master` branch and retest?

Regarding initialization process: it is done completely on device side and is independent. The only way I am aware of to interrupt it is to cut the power. Nitrokey App could be closed and reopened while this progresses.

alem0lars, 2017-10-31:
*   Storage firmware version: 0.48
*   Nitrokey App version: 1.11
* OS: Gentoo 
*   kernel version: Linux kate 4.12.12-gentoo #1 SMP Sat Sep 30 07:21:12 CEST 2017 x86_64 Intel(R) Core(TM) i7-4710MQ CPU @ 2.50GHz GenuineIntel GNU/Linux

szszszsz, 2018-02-16:
Hi @alem0lars !
With v0.49/v0.50 smartcard synchronization is more strict. This may clear issues on setups, where smartcard is used frequently through CCID interface. Could you let us know on this page is it still occurring for you?
@alex-nitrokey Have you reproduced this issue?

alex-nitrokey, 2018-02-19:
Well I am not sure anymore, but I think I would have mentioned it here 🤔 I probably hadn't the right environment at hand back then.

Let's see what @alem0lars is reporting, before I start to investigate a may solved issue.