Nitrokey/nitrokey-storage-firmware #2: AES keys regeneration should not force SD card random fill

szszszsz, 2016-05-25:
Hi!
Currently AES keys regeneration command sets `SDFillWithRandomChars_u8` to `0` which forces user to make SD card random fill each time.
SD card random overwrite should be done only on first use of given SD card and should not be connected to AES keys regeneration. 

For more details please check: https://github.com/Nitrokey/nitrokey-app/issues/45#issuecomment-220589776 and https://github.com/Nitrokey/nitrokey-app/issues/45#issuecomment-221548662

Attaching logs: [after_generating_new_AES_keys.log.txt](https://github.com/Nitrokey/nitrokey-storage-firmware/files/281864/after_generating_new_AES_keys.log.txt)
[before_generating_new_AES_keys.log.txt](https://github.com/Nitrokey/nitrokey-storage-firmware/files/281863/before_generating_new_AES_keys.log.txt)

rudbo, 2016-06-13:
Firmware side done
https://github.com/Nitrokey/nitrokey-storage-firmware/commit/5fea81b72d95de0aadae7c69fd76d9c37df754c2

jans23, 2016-06-13:
Thank you. Does this change require a modification of the App as well?

>

szszszsz, 2016-06-13:
@jans23 @rudbo 
It looks like I have prepared a fix some time ago in the branch `issue_45_3-initialize_device`.
https://github.com/Nitrokey/nitrokey-app/compare/issue_45_3-initialize_device
Please check.

szszszsz, 2016-06-22:
Works as designed, closing.

closed 2016-06-22 by szszszsz