Nitrokey/nitrokey-storage-firmware #3: Enabling hidden volume results in stick freeze

szszszsz, 2016-06-07:
Stick appears to be halted after enabling hidden volume in Nitrokey App and checking partitions using parted. Green Led stayed on. Application reported sticks connection and disconnection mutliple times. Attaching log.

Occurence: once
Nitrokey Storage 0.34
Ubuntu 16.04
App https://github.com/Nitrokey/nitrokey-app/commit/0dd16309609428436b85808b9e22ad41328b2e4d
[hidden-volume-issue.txt](https://github.com/Nitrokey/nitrokey-storage-firmware/files/302782/hidden-volume-issue.txt)

Scenario:
1. Unlock encrypted volume
2. Create partition table and format as fat32
3. Unlock hidden volume
4. Create partition table and format as fat32
5. Lock device
6. Unlock encrypted and hidden device (Unlock of hidden device is unsuccessful)
7. Create another hidden volume in different slot with same password
8. Unlock hidden volume
9. Check partition list with `sudo parted /dev/sdc` and command `print`

closed 2016-06-25 by jans23

szszszsz, 2016-06-25:
Tested multiple hidden volume slots on 0.35 and Ubuntu 16.04. Issue has not occured.

jans23, 2016-06-25:
Great. Thanks.
-- 
Sent from a mobile device.