Nitrokey/nitrokey-storage-firmware #5: Reboot command

szszszsz, 2016-06-10:
I would make use of command which reboots stick. Currently I have to remove and insert again same device for achieving same 'start' state. Would that be possible to implement? Or, is there any other way?
I thought it could be possible through `dfu-programmer at32uc3a3256s start`, but it shows error it could not find any device.

jans23, 2016-06-10:
@szszszsz Please elaborate what you mean with start state and what do you want to achieve actually.

szszszsz, 2016-06-10:
@jans23 I would like to achieve using command same device state as it is in just after inserted to USB port. A command which would 'reboot' the OS on stick.

jans23, 2016-06-10:
@szszszsz What in your scenario is the different to the start stage? For instance, would it be sufficient to lock the device (and reverse that PINs have been entered)?

szszszsz, 2016-06-10:
@jans23 Particularly I would like to access to the unencrypted volume in case when stick 'removes the medium' from it. Reinserting the device solves the problem. Software solution would be better though.

jans23, 2016-06-10:
@szszszsz Would a [USB port reset](https://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line) be an option?

szszszsz, 2016-06-10:
@jans23 It would be great if this would work! I have tried sending messages to kernel but with no success. I will check it the source code.

szszszsz, 2016-06-13:
Unfortunately, your second link is not working. About first one - looking at dmesg it sends reset command to controller, but device do not look like reset. 
I need the reboot command  to workaround https://github.com/Nitrokey/nitrokey-storage-firmware/issues/6.

rudbo, 2016-06-13:
New HID cmd
CMD_RESET_STICK - 0x71, called with user password

https://github.com/Nitrokey/nitrokey-storage-firmware/commit/86d2aece982c9a180c36e3f47dc0ecac71a53cb6

jans23, 2016-06-29:
@szszszsz Can you confirm that you don't need the reboot command anymore?

szszszsz, 2016-06-29:
Currently I do not need it. It was only to avoid `prodinfo` side effects, but it was worked around anyway. It might be reverted for now and restored in case it would be needed to avoid any other side effects (but only for quick workarounds). It might be left as well since it does not look as security threat.

I have not added implementation of sending this command to application until now.

szszszsz, 2016-06-29:
Since it has been added to firmware, this issue could be safely closed. I have left it open to test the command but had not time unfortunately.

jans23, 2016-06-29:
I suggest to remove this feature to reduce the potential of any unforeseen sideeffects.

rudbo, 2016-06-29:
Command in firmware removed
https://github.com/Nitrokey/nitrokey-storage-firmware/commit/ccb7554ad57265eecaaf7c4bcc712e496a91b3d0

closed 2016-06-29 by jans23