Nitrokey/nitrokey-storage-firmware #6: prodinfo command removes unencrypted volume from system

szszszsz, 2016-06-13:
Occurency: always
Ubuntu 16.04
Nitrokey Storage
Prodinfo string:
`CPU:0xf4780feb,SC:0x000000dc,SD:0x035503a4,SCM:0x1b,SCO:0x534d,DAT:2012.14,Speed:8820,Size:8,Firmware:0.34 - 5`
Nitrokey app sha# https://github.com/Nitrokey/nitrokey-app/commit/0dd16309609428436b85808b9e22ad41328b2e4d

After issuing:
```
./nitrokey-app --cmd prodinfo
```
unencrypted volume disappears from the system and green LED blinks all the time. 
Device physical removal and reinsertion restores correct functionality.

Attached parted output: `parted -l`
[parted_before.txt](https://github.com/Nitrokey/nitrokey-storage-firmware/files/311802/parted_before.txt) [parted_after.txt](https://github.com/Nitrokey/nitrokey-storage-firmware/files/311803/parted_after.txt)
And `dmesg`: [dmesg.txt](https://github.com/Nitrokey/nitrokey-storage-firmware/files/311812/dmesg.txt)

rudbo, 2016-06-15:
This command is only for productions test, problems with normal usage could be

jans23, 2016-06-15:
@szszszsz Is this function working good enough for your factory tool?

szszszsz, 2016-06-15:
Hi @jans23 @rudbo !
Because of this bug it is not possible to do any operations connected to unencrypted volume after getting `prodinfo` string without physical intervention - removing and inserting the same device. Therefore sequential tests are not possible. This also the only function which identifies fully the stick (giving serial numbers and storage size), so it has to be called each time. While it gives only information it should not make any side-effects.

Is there a possibility to create similar function with trimmed `prodinfo` (e.g. with problematic fields missing) which would avoid side effects?

jans23, 2016-06-15:
@szszszsz Rudolf checked in a new firmware which includes the reboot function. Would it be a workaround to reboot the device after getting prodinfo?

szszszsz, 2016-06-15:
@jans23 I will check tomorrow and let you know if it helped.

jans23, 2016-06-24:
This issue is still valid with firmware 0.35 and latest App.

rudbo, 2016-06-25:
Remove sd write test from prodinfo command
Prodinfo with SD write test is an new cmd: 
#define STICK20_CMD_PRODUCTION_TEST_WITH_WRITE_TEST     (STICK20_CMD_START_VALUE + 27)

Fixed with https://github.com/Nitrokey/nitrokey-storage-firmware/commit/cb4ef7d1863f993946db8937f122457dbe498a82

szszszsz, 2016-06-25:
Device is now not disappearing after command execution. 
Checked with https://github.com/Nitrokey/nitrokey-storage-firmware/commit/79e07a71779d712ef832bbeac717a5cd66278761 and https://github.com/Nitrokey/nitrokey-app/commit/0a89ca53908552e2edcd353381124197c992807d

closed 2016-06-25 by szszszsz