Nitrokey/nitrokey-storage-firmware #7: prodinfo command never works first time

szszszsz, 2016-06-13:
Occurency: always
Ubuntu 16.04
Nitrokey Storage
Prodinfo string:
CPU:0xf4780feb,SC:0x000000dc,SD:0x035503a4,SCM:0x1b,SCO:0x534d,DAT:2012.14,Speed:8820,Size:8,Firmware:0.34 - 5
Nitrokey app sha# Nitrokey/nitrokey-app@0dd1630

Command:
```
./nitrokey-app --cmd prodinfo
```
never works first time. Running second time works.

szszszsz, 2016-06-17:
Hi @rudbo !
Could you please check is device `prodinfo` command called as designed in application?
https://github.com/Nitrokey/nitrokey-app/blob/master/src/ui/mainwindow.cpp#L644

rudbo, 2016-06-18:
Wait for stick response in prodinfo function
https://github.com/Nitrokey/nitrokey-app/commit/6d9e6952752012e162f85207608db3890c039c34

jans23, 2016-06-24:
This issue is still valid for me. I assume it is because the Production Tool queries the smart card right before and the firmware locks it for several seconds.

rudbo, 2016-06-25:
Now prodinfo only get firmware version and SD card infos, no smartcard access
https://github.com/Nitrokey/nitrokey-storage-firmware/commit/bc53ce3596acddc516933691b87970ea70ba30dc

szszszsz, 2016-06-25:
This works great! I have checked it 3 times in in production tool and each worked first time.
Checked with 0.36 https://github.com/Nitrokey/nitrokey-storage-firmware/commit/79e07a71779d712ef832bbeac717a5cd66278761 and https://github.com/Nitrokey/nitrokey-app/commit/0a89ca53908552e2edcd353381124197c992807d

closed 2016-06-25 by szszszsz