Nitrokey/nitrokey-storage-firmware #9: Initiating keys with wrong password results in flag StickKeysNotInitiated change

szszszsz, 2016-06-24:
StickKeysNotInitiated is changing after sending command `STICK20_CMD_GENERATE_NEW_KEYS` to the device with incorrect admin password.

Prereq:
Nitrokey Storage, v0.35
Ubuntu 16.04

Scenario:
1. Flash device
2. Make factory reset with application `nitrokeyapp --cmd factoryReset`
3. Run Nitrokey Application (in `debug` mode)
4. Select `Initialize device` from application context menu
5. Accept keys generation
6. Enter incorrect password for Admin, like `123123123`
7. Accept further messages (click `Ok`), but do not agree for SD card clearing
8.  `Initialize device` menu item is not shown (since it is depending on `StickKeysNotInitiated` flag) 

Keys should not be generated (since incorrect password has been passed) but the flag `StickKeysNotInitiated` is changing. Please check is that expected behavior.

Before scenario:
```
GetStick20Configuration
HID_GetStick20Configuration
MagicNumber_StickConfig_u16      : 12824
ReadWriteFlagUncryptedVolume_u8  : 0
ReadWriteFlagCryptedVolume_u8    : 0
VersionInfo_au8[4]               : 0 35 0 6
ReadWriteFlagHiddenVolume_u8     : 0
FirmwareLocked_u8                : 0
NewSDCardFound_u8                : 1
SDFillWithRandomChars_u8         : 0
ActiveSD_CardID_u32              : 0x035503A4
VolumeActiceFlag_u8              : 1
NewSmartCardFound_u8             : 0
UserPwRetryCount                 : 3
AdminPwRetryCount                : 3
ActiveSmartCardID_u32            : 0x3B60
StickKeysNotInitiated            : 1
```

After scenario:
```
GetStick20Configuration
HID_GetStick20Configuration
MagicNumber_StickConfig_u16      : 12824
ReadWriteFlagUncryptedVolume_u8  : 0
ReadWriteFlagCryptedVolume_u8    : 0
VersionInfo_au8[4]               : 0 35 0 6
ReadWriteFlagHiddenVolume_u8     : 0
FirmwareLocked_u8                : 0
NewSDCardFound_u8                : 1
SDFillWithRandomChars_u8         : 0
ActiveSD_CardID_u32              : 0x035503A4
VolumeActiceFlag_u8              : 1
NewSmartCardFound_u8             : 0
UserPwRetryCount                 : 3
AdminPwRetryCount                : 2
ActiveSmartCardID_u32            : 0x3B60
StickKeysNotInitiated            : 0
```

Same after both application restart and device reinsertion:

```
GetStick20Configuration
HID_GetStick20Configuration
MagicNumber_StickConfig_u16      : 12824
ReadWriteFlagUncryptedVolume_u8  : 0
ReadWriteFlagCryptedVolume_u8    : 0
VersionInfo_au8[4]               : 0 35 0 6
ReadWriteFlagHiddenVolume_u8     : 0
FirmwareLocked_u8                : 0
NewSDCardFound_u8                : 1
SDFillWithRandomChars_u8         : 0
ActiveSD_CardID_u32              : 0x035503A4
VolumeActiceFlag_u8              : 1
NewSmartCardFound_u8             : 0
UserPwRetryCount                 : 3
AdminPwRetryCount                : 2
ActiveSmartCardID_u32            : 0x3B60
StickKeysNotInitiated            : 0
```

szszszsz, 2016-06-24:
^^^^ updated scenario - divided to initialization and test scenario to make description more clear

rudbo, 2016-06-25:
Fixed with
https://github.com/Nitrokey/nitrokey-storage-firmware/commit/411fc36fe458cb6748470fffd45836f82c372a56

szszszsz, 2016-06-25:
Will be checked on next binary release.

szszszsz, 2016-06-25:
It works on Ubuntu 16.04 and
https://github.com/Nitrokey/nitrokey-storage-firmware/commit/79e07a71779d712ef832bbeac717a5cd66278761
https://github.com/Nitrokey/nitrokey-app/commit/0a89ca53908552e2edcd353381124197c992807d

closed 2016-06-25 by szszszsz