Nitrokey/nitrokey-storage-firmware #10: No serial numbers presented on prodinfo results

szszszsz, 2016-06-25:
One of my sticks suddenly stopped sending serial numbers. This is an excerpt from `prodlog.txt`:

```
timestamp:25.06.2016 13:21:21,CPU:0x00000000,SC:0x00000000,SD:0x035503a4,SCM:0x1b,SCO:0x534d,DAT:2012.14,Size:8,Firmware:0.36 - 7
```
Full file: [prodlog.txt](https://github.com/Nitrokey/nitrokey-storage-firmware/files/333050/prodlog.txt)

I am checking will second stick behave same after firmware upgrade.

szszszsz, 2016-06-25:
My other stick (firmware 0.35) works with https://github.com/Nitrokey/nitrokey-app/commit/0a89ca53908552e2edcd353381124197c992807d
```
Get production infos
Firmware     0.35 - 6
CPU ID       0xf4780feb
Smartcard ID 0x00003b60
SD card ID   0x035503a4

SD card infos
Manufacturer 0x1b
OEM          0x534d
Manufa. date 2012.14
Size          8 GB
```

After firmware upgrade to 0.36 the serials are gone:
```
Firmware     0.36 - 7
CPU ID       0x00000000
Smartcard ID 0x00000000
SD card ID   0x035503a4

SD card infos
Manufacturer 0x1b
OEM          0x534d
Manufa. date 2012.14
Size          8 GB

```

szszszsz, 2016-06-25:
Card serial number (SC) is shown correctly however in application's `About` dialog.

szszszsz, 2016-06-29:
@jans23 I think this might be important issue, since without serial numbers it will be impossible to identify broken devices and assign logs information from production process.

jans23, 2016-06-29:
The latest firmware includes the CPU ID. (The SC ID is still nulled but that should be ok). @szszszsz Please test.

szszszsz, 2016-06-29:
Current `prodinfo` output:

```
sz@feather:~/work/nitrokey-app⟫ ./nitrokey-app --cmd prodinfo
Application started successfully.
Connecting to nitrokey. connected. 
Send -get production infos-

Get production infos
Firmware     0.37 - 7
CPU ID       0xf4780feb
Smartcard ID 0x00000000
SD card ID   0x035503a4

SD card infos
Manufacturer 0x1b
OEM          0x534d
Manufa. date 2012.14
Size          8 GB

Stick OK
```
@jans23 Will Smartcard ID be needed? If not, then I can close the issue.

closed 2016-06-29 by jans23