Nitrokey/nitrokey-storage-firmware #11: Firmware password change possible only once

szszszsz, 2016-06-27:
Discovered while working on issue https://github.com/Nitrokey/nitrokey-app/issues/97
It looks like device returns success only on first correct change (=> first change after factory reset, device reinsertion and application restart are not helping). Later password changes are always reported by device as failed.
Tested on both Windows 8.0 (compiled with Qt's mingw 5.3) and Ubuntu 16.04. I was using branch `issue_97_firmware_password_feedback` (the only differences in code with mainline are debug prints and feedback logic).

Prereq:
Firmware 0.36
App https://github.com/Nitrokey/nitrokey-app/commit/bc3ce95ab25eef7a42aa728b5d8a31359b986eb3#diff-3301dce0d4fa3f36ee73aa1a548416e0R2401
Both Ubuntu and Windows
Device is just after factory reset

Scenario:
1. Run application `./nitrokey-app -d -a`
2. Initialize device (generate keys but do not clear SD, disable warning)
3. Change firmware password to `asdasdasd` using as current `12345678`
4. Repeat step 3 with current `asdasdasd` and new one `123123123` (should work but it does not) 

Attaching output from app and debug log.
[issue_97.txt](https://github.com/Nitrokey/nitrokey-storage-firmware/files/335740/issue_97.txt)
[issue_97_debug_log.txt](https://github.com/Nitrokey/nitrokey-storage-firmware/files/335735/issue_97_debug_log.txt)

As for App output clarification:
```
Old:  "12345678"
New1:  "asdasdasd"
New2:  "asdasdasd"
Old: "12345678"
New: "asdasdasd"
```
The last two lines (which are printed from function sending command to device) are just confirmation of the first 3 lines (which is Qt's form output). In case it would be confusing please let me know so I could simplify it.

jans23, 2016-06-28:
@szszszsz I can change the firmware password without problems. Is this issue still appearing for you?

jans23, 2016-06-28:
@szszszsz Could you provide the debug log created by the App, please?

szszszsz, 2016-06-28:
Hi @jans23 
Debug log is attached in issue description: https://github.com/Nitrokey/nitrokey-storage-firmware/files/335735/issue_97_debug_log.txt

I can change the password but only first time (I have `success` feedback from device only once).
Could you share test scenario which is working for you?

jans23, 2016-06-28:
I use firmware v0.36 andd latest App from master branch to change the firmware password 12345678 -> 87654321 -> 11111111 -> 22222222 -> "password", enabled the firmware update, flashed v0.36, changed firmware password from "password" -> 12345678.

System: Ubuntu 16.04

szszszsz, 2016-06-28:
@jans23 I confirm your scenario works correctly on my device. I see however that all your password have 8 characters (mine had 9), is there any constraint set on firmware password?

szszszsz, 2016-06-28:
@jans23 My scenario is still not working: 12345678 -> asdasdasd -> 123123123

jans23, 2016-06-28:
Now I can reproduce the issue:

1. Change: 12345678 -> 0987654321 - OK
2. Change back: 0987654321 -> 12345678 - ERROR
3. Change back: 09876543 -> 12345678 - OK

So I guess all characters more than 8 are dropped somewhere silently.

szszszsz, 2016-06-28:
Should firmware password length be increased then or should constraints for length be added in password dialog along with description?

jans23, 2016-06-28:
The App should check the password/PIN length and deal properly with too long passwords/PINs. This applies for User PIN, Admin PIN and Firmware Password.

But the Firmware Password's length must not be limited to 8 characters! @rudbo What is the maximum length from the firmware's perspective?

szszszsz, 2016-06-28:
@jans23 Application currently is not letting user to input passwords (of any type) longer than 20 characters. Also information about it is in password dialog description, under the fields.

jans23, 2016-06-28:
That should be correct. Lets wait for rudbo's answer tomorrow.

rudbo, 2016-06-29:
The problem was that that length of the old password was taken as the length of the new password
https://github.com/Nitrokey/nitrokey-storage-firmware/commit/1c0f563c13f82e941d9b9a6ca758bf2158c5dfa7
Thanks for your help. 

The lenght of the firmware password is only max 15 byte, because on the stick a salt of 10 byte is added to the firmware password.

szszszsz, 2016-06-29:
Test scenario contained in issue description now works correctly. However I cannot set password with 15 characters. It works correctly for 14 characters though. Please check.
The passwords I have used:
12345678 -> 12345678901234 -> 123456789012345 (fail)

szszszsz, 2016-06-29:
The 14 char limit might be caused by implementation in Nitrokey App, but it looks correctly:
```
// CS20_MAX_UPDATE_PASSWORD_LEN = 15
 n = strlen((const char *)new_password);
  if (CS20_MAX_UPDATE_PASSWORD_LEN <= n) {
    return (false);
  }
```

szszszsz, 2016-06-29:
After changing `CS20_MAX_UPDATE_PASSWORD_LEN <= n` to `CS20_MAX_UPDATE_PASSWORD_LEN < n` (which allowed to use 15 chars password) device is rejecting it. So it looks like the limit is set currently in firmware to 14 chars.

jans23, 2016-06-29:
@szszszsz I believe this issue and #12 are actually the same because entering factory update mode is just a consequence. If you agree, please close #12.

szszszsz, 2016-06-29:
@jans23 It was initially, but currently it looks like (in issue #12) it is not possible to use 14 chars password to enter update mode. Would you like to report it as another issue?

jans23, 2016-06-29:
No, thats fine.

rudbo, 2016-06-29:
Change length of firmware pin to 20 char
https://github.com/Nitrokey/nitrokey-storage-firmware/commit/ccb7554ad57265eecaaf7c4bcc712e496a91b3d0

szszszsz, 2016-06-29:
@jans23 Please do not check firmware password change on firmware 0.38 until proper changes on application will be applied.

szszszsz, 2016-06-29:
@jans23 If possible I would ask to revert change which disables reseting firmware password during factory reset (or better - prepare aside special version for development) for a time of applying proper changes to application.

szszszsz, 2016-07-01:
Changing firmware password to new one with 20 chars now works correctly. 
Tested on Ubuntu 16.04 and firmware 0.38 with https://github.com/Nitrokey/nitrokey-app/commit/c8ea4064f2a6dc2b710efa78e024eeb98d688542

closed 2016-07-01 by szszszsz