Nitrokey/nitrokey-storage-firmware #12: Device is not accepting firmware password after changing

szszszsz, 2016-06-27:
Discovered while working on issue https://github.com/Nitrokey/nitrokey-app/issues/97
Might be connected with #11 

Prereq:
Firmware 0.36
App https://github.com/Nitrokey/nitrokey-app/commit/bc3ce95ab25eef7a42aa728b5d8a31359b986eb3#diff-3301dce0d4fa3f36ee73aa1a548416e0R2401
Ubuntu 16.04. I was using branch `issue_97_firmware_password_feedback` (the only differences in code with mainline are debug prints and feedback logic).

Device has just changed firmware password with application from `12345678` to `asdasdasd`

Scenario:
1. Send update command `./nitrokey-app --cmd setUpdateMode=asdasdasd` 
2. Check is device in update mode

szszszsz, 2016-06-29:
Device is not entering update mode with 14 characters length long password (with proper fixes in Nitrokey app, which will be merged in a moment). @rudbo please check `STICK20_CMD_ENABLE_FIRMWARE_UPDATE` command for password length limit.

rudbo, 2016-06-29:
Change length of firmware pin to 20 char
https://github.com/Nitrokey/nitrokey-storage-firmware/commit/ccb7554ad57265eecaaf7c4bcc712e496a91b3d0

szszszsz, 2016-07-01:
Setting update mode with 20 chars firmware password now works correctly. 
Tested on Ubuntu 16.04 and firmware 0.38 with https://github.com/Nitrokey/nitrokey-app/commit/c8ea4064f2a6dc2b710efa78e024eeb98d688542

closed 2016-07-01 by szszszsz