Nitrokey/nitrokey-storage-firmware #17: TOTP not working for time bigger than int32

szszszsz, 2016-10-17:
TOTP returns wrong code with time set bigger than `int32` max value (`int32_max = 2 ** 31 - 1`) according to both RFC spec and handcrafted test. NK Pro passes this. Attaching full log from test run: 
[TOTP_64bit_issue_log.txt](https://github.com/Nitrokey/nitrokey-storage-firmware/files/534275/TOTP_64bit_issue_log.txt) (most interesting part is at the end, but the rest might be helpful too). 

Date for max `int32` is: 
```
In [42]: datetime.datetime.fromtimestamp(2**31-1)                                                           
Out[42]: datetime.datetime(2038, 1, 19, 4, 14, 7)
```

szszszsz, 2019-06-14:
A. Storage v0.53+ (to be released as v0.54; current master):
1. Time is taken from the internal clock. No obvious limitation is seen here. In common Unix time() function doc [4] it is mentioned, that some systems may have it 31 bits.
2. The window id is calculated on both 64 bits numbers.
3. Full 64 bit value is sent to hmac-sha1 calculation.

Tests show, that Storage is not passing over 2^31 border for time.
Perhaps the internal clock is signed 32 bit. As registered in issue [8] - to correct. The cause is probably the lines 1211-1212, where the time is taken from the clock, instead of taking it from the App, and adding internal time as a relative value.

[1] https://github.com/Nitrokey/nitrokey-storage-firmware/blob/master/src/OTP/hotp.c#L1212
[2] https://github.com/Nitrokey/nitrokey-storage-firmware/blob/master/src/OTP/hotp.c#L1219
[3] https://github.com/Nitrokey/nitrokey-storage-firmware/blob/master/src/OTP/hotp.c#L1227
[4] (...) even on ABIs where time_t is a signed 32-bit integer and the
clock ticks past the time 2**31 (2038-01-19 03:14:08 UTC, ignoring leap
seconds) (...)
[8] https://github.com/Nitrokey/nitrokey-storage-firmware/issues/17

szszszsz, 2019-06-18:
Fix proposition:
We do not need to set the system clock to the actual, absolute time, but instead it can count relative seconds from the start. This way we extend the range to current/any time + 68 years (the 32-bit range of  the clock), instead of now implemented 1970+68 years.

In pseudo-code:

A. Current implementation:
```C
u32 dev_time = 0; // increments with every second, regardless of value
dev_time = time_from_the_nkapp; // still u32
otp_code = get_code(dev_time); // get_code accepts u64, but gets u32
```

B. Proposed impl.:
```C
u32 dev_time = 0;  // increments with every second, regardless of value
u64 nkapp_time = time_from_the_nkapp;
otp_code = get_code(nkapp_time + dev_time); // sum of both is u64
```

szszszsz, 2019-06-18:
@NKelias Could you take a look on it?

NKelias, 2019-06-18:
Passes https://github.com/Nitrokey/libnitrokey/blob/0275ebc610b255e3519baf866438244f288576bc/unittest/test_pro.py#L456 now.

Excellent description, by the way! :1st_place_medal:

PR #90 (2019-06-18): Use system clock for local offset, not timekeeping
This uses the system clock for storing the offset since the last update only.
Unix timestamp of the `set_time` command is stored locally as `u64` to prevent year 2038 integer overrun and added to current local offset for TOTP generation.

resolves #17 

Firmware: 
[firmware.zip](https://github.com/Nitrokey/nitrokey-storage-firmware/files/3301984/firmware.zip)
commit baaf0467 (2019-06-18): Merge branch 'pr_90'

Allow TOTP implementation to work with timestamps beyond 2038

From #90:
> This uses the system clock for storing the offset since the last update only.
Unix timestamp of the set_time command is stored locally as u64 to prevent year 2038 integer overrun and added to current local offset for TOTP generation.
> Passes https://github.com/Nitrokey/libnitrokey/blob/0275ebc610b255e3519baf866438244f288576bc/unittest/test_pro.py#L456 now.

Fixes #17
Fixes #90
closed 2019-06-18 by szszszsz