Nitrokey/nitrokey-storage-firmware #18: exported firmware doesn't match imported firmware

jans23, 2016-10-28:
Perhaps because of a different bootloader...

szszszsz, 2017-05-15:
An example of the difference from our support forum:
https://support.nitrokey.com/t/nitrokey-storage-firmware-checksum-mismatch/455

szszszsz, 2017-05-15:
Apparently the issue is connected with different state of cleared bit on mediums - `1` on flash and `0` usual. The exported firmware have `1`'s (instead of `0`'s in original file) at the file end. More details in forum link above.

jans23, 2017-08-03:
@szszszsz An idea is to provide a separate script which converts the exported firmware so that it matches the initial firmware. Do you think this is a reasonable solution?

szszszsz, 2017-08-04:
Actually the issue is to differentiate the empty, not used flash space from the `1`'s series written in firmware file. It could be achieved by adding the magic bytes values at the end of both bootloader's and actual firmware's code (perhaps in linker script). With such marks it would be trivial to replace flashes default bit values `1` to `0` and make the firmware's same. I do not think it is possible to do with current exported data.

However, at current state I can provide a tool that matches the exported and genuine firmware by skipping the `1` and `0` differences found at the end of the blocks.

It is worth mentioning (this might not be obvious because of the difference mentioned in issue) that exported firmware is usable to flash back to the device and it would result in same flash code state.

jans23, 2017-08-04:
We could leave it to Alex to provide such tool.

rudbo, 2017-08-25:
To add 0xFF to the flashed firmware file use the command 
srec_cat firmwareVXXX.bin --Binary --offset=0x00000000 -fill 0xFF 0x00000 0x3E000 --output firmwareVXXX_filled.bin --Binary

Download srec_cat under https://sourceforge.net/projects/srecord/files/srecord-win32/
Doc http://srecord.sourceforge.net/man/man1/srec_cat.html

adlererik, 2017-12-11:
This almost replaces all the 0s to 1s. Enough so that it is easy to verify the binaries. There are still a few small regions that are displayed as 0s. Is there a work around? 

$ aptitude show srecord |grep Ver
Version: 1.58-1.1+b2

$ srec_cat firmware_V0.48.0.bin --Binary --offset=0x00000000 -fill 0xFF 0x00000 0x3E000 --output firmware_V0.48.0_filled.bin --Binary

$ diff  <(xxd firmware_V0.48.0_filled.bin) <(xxd firmware.bin)

11291,11296c11291,11296
< 0002c1a0: 1639 f7dc eb0b c77b 0000 0000 0000 0000  .9.....{........
< 0002c1b0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
< 0002c1c0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
< 0002c1d0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
< 0002c1e0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
< 0002c1f0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
---
> 0002c1a0: 1639 f7dc eb0b c77b ffff ffff ffff ffff  .9.....{........
> 0002c1b0: ffff ffff ffff ffff ffff ffff ffff ffff  ................
> 0002c1c0: ffff ffff ffff ffff ffff ffff ffff ffff  ................
> 0002c1d0: ffff ffff ffff ffff ffff ffff ffff ffff  ................
> 0002c1e0: ffff ffff ffff ffff ffff ffff ffff ffff  ................
> 0002c1f0: ffff ffff ffff ffff ffff ffff ffff ffff  ................
11632c11632
< 0002d6f0: 0000 007d 0000 0000 0393 8700 0393 8700  ...}............
---
> 0002d6f0: 0000 007d ffff ffff 0393 8700 0393 8700  ...}............

techge, 2017-12-14:
We could create a tool which compares the exported firmware with another and ignores the difference of the zeros and ones. I am just wondering if this is so much better than comparing it manually... 

But wouldn't it be better to fix it on firmware/app/whatever level, so that it exports a comparable version in the first place? I admit that I have not a specific idea how difficult it would be to achieve this instead of creating a simple tool. But it feels like this approach is more trustworthy and is easier for users.

adlererik, 2017-12-18:
Since the addresses of the offending 0s are known it would be relatively easy to write a tool/template that just inverts the offending 0s to 1s.  A bit of a hack but would probably work.

jadeaffenjaeger, 2017-12-26:
I also don't see a good way of implementing this in firmware without massively interfering with the actual memory dump. When blanking out the ones with a script, it should also be possible to not use a magic byte value, but instead have the linker generate a map file and extract the relevant memory regions from that.

szszszsz, 2018-02-16:
On recent firmware's [release page](https://github.com/Nitrokey/nitrokey-storage-firmware/releases/tag/0.50) I saw a binary with `_filled` suffix - is this meant to be comparable with a dump from the application?

alex-nitrokey, 2018-02-19:
It isn't. At least the behaviour seems to be the same with 0.50. 
Maybe @rudbo has a idea, what this is about :-)

LouSparfell, 2018-05-07:
Any news about this ? How I know if my firmware has been backdoored or not ? The checksum is verified in the nitrokey-app when it displays the version or not ? Or you still need to verify it manually ?

szszszsz, 2018-05-15:
Hi @LouSparfell ! Nothing new I am afraid. I would go with https://github.com/Nitrokey/nitrokey-storage-firmware/issues/18#issuecomment-350611808 at the moment if I would like to check the firmware.

Another idea, only if you could afford to lose OTP/PWS data on the device, is to:
- dump firmware and copy it to local disk,
- flash the device with a official release and same version,
- dump firmware and compare with the previous one.

This should result in exactly identical dumps.