Nitrokey/nitrokey-storage-firmware #19: Export firmware command saves data on unencrypted storage

szszszsz, 2016-11-03:
Export firmware command saves data on unencrypted storage making possibly a security risk since it could not be easily securely wiped-out from this volume. In case attacker steals device and identifies firmware using found data on unencrypted volume he can attack using known vulnerabilities if firmware is not up to date.

jans23, 2016-11-03:
The source code is published at Github and the firmware version can be
requested via the HID interface of the device (its shown in the About
window already). Since the exported firmware doesn't contain user data I
don't see a problem.

>

szszszsz, 2016-11-03:
Indeed firmware version is already easily accessible, fair point.
Even if it would not be known, an adversary could simply run all his malicious exploits until some would hit. I rather overrated this information after all.

closed 2016-11-03 by szszszsz