Nitrokey/nitrokey-storage-firmware #24: `No TDs queued` messages on USB3

szszszsz, 2016-12-01:
Users reports about dmesg messages similar to the one below while using USB3:
```
[ 578.959764] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 2 ep 10 with no TDs queued?
```

szszszsz, 2017-01-21:
To check on v0.44 is this message still appears in logs.

szszszsz, 2017-08-04:
To check on v0.46

szszszsz, 2017-08-04:
Message is shown just after invoking `gpg2 --card-status`:
```
xhci_hcd 0000:00:14.0: WARN Event TRB for slot 16 ep 10 with no TDs queued?
```
Ubuntu 16.10
Storage 0.46.5

szszszsz, 2017-08-17:
No messages seen on 0.47.1

szszszsz, 2017-08-23:
Occurs on v0.48.  `Warn` messages are still shown in system journal when invoking `gpg2 --card-status`.
```
kernel: xhci_hcd 0000:00:14.0: WARN Event TRB for slot 28 ep 10 with no TDs queued? 
```

rudbo, 2017-08-25:
The problem comes from the kernel usb lib. The kernel usb lib want to send a request to endpoint 10.
But the storage has only endpoints 0-6, so that is normal that there is no transfer descriptor (TD)
Why the kernel or gpg2 generate this request is unknown.

On the outgoing usb side there is no unusual usb packet send to the stick, so I think that there is no problem to fix on the firmware side.

TRB  = Transfer Request Block 
TD = Transfer Descriptor

closed 2017-08-25 by rudbo