Nitrokey/nitrokey-storage-firmware #25: smartcard.Exceptions.CardConnectionException: Unable to connect with protocol: T0 or T1. Sharing violation

Germano0, 2017-03-01:
` $ python2 tool/gnuk_remove_keys.py -p`
returns
```
Admin password: 
Traceback (most recent call last):
  File "tool/gnuk_remove_keys.py", line 112, in <module>
    main(passwd)
  File "tool/gnuk_remove_keys.py", line 86, in main
    gnuk.connection.connect()
  File "/usr/lib64/python2.7/site-packages/smartcard/CardConnectionDecorator.py", line 54, in connect
    self.component.connect(protocol, mode, disposition)
  File "/usr/lib64/python2.7/site-packages/smartcard/pcsc/PCSCCardConnection.py", line 128, in connect
    SCardGetErrorMessage(hresult))
smartcard.Exceptions.CardConnectionException: Unable to connect with protocol: T0 or T1. Sharing violation.
```

`python-2.7.13-1.fc25.x86_64` `pyscard-1.9.0-5.fc25.x86_64` on Fedora 25

jans23, 2017-03-02:
Do you have the latest firmware (v.045) installed?

Germano0, 2017-03-02:
@jans23 Nitrokey App says firmware 0.8

jans23, 2017-03-02:
It seems there is some confusion about the different Nitrokey models: You have a Nitrokey Pro but you followed instructions for Nitrokey Start and posted this issue in the project for Nitrokey Storage. If you want to delete keys on a Nitrokey Pro, you may want to [reset the device](https://www.nitrokey.com/de/documentation/frequently-asked-questions#how-to-reset-a-nitrokey) which deletes all three keys (and all other settings).

I'm closing this issue now. If you still have problems, please post a message in our [support forum](http://support.nitrokey.com/).

closed 2017-03-02 by jans23