Nitrokey/nitrokey-storage-firmware #27: ReadMinWatermark > ReadMaxWatermark

szszszsz, 2017-03-07:
Windows
Storage v0.43
Nitrokey App v0.6.3

While setting up hidden volume application has shown that it can be placed between `50`th and `49`th percentile which is not possible. These two values are shown directly as the device has sent to the Application (except it is x<10 or x>90). It would be great to add proper check in firmware so the watermarks values could be in proper relation to each other.

Details: https://support.nitrokey.com/t/hidden-volume-and-plausible-deniability/388/3

jans23, 2017-03-20:
The NK Storage records only the read watermarks, he does not use them any further. The restriction is made in the NK App, perhaps you should only give a warning if the values ​​are critical and then do not completely prevent it.

Why in the case 49/50 percent is displayed can explain as follows: The min values ​​are limited to 50% of the SD block number on the storage, during the transfer to the app percentages are calculated, which then are the SD block number + -1 for rounding reasons:

```
    // +-1 needed by rounding cuts
    output[OUTPUT_CMD_RESULT_OFFSET + DEVICE_HIGH_WATERMARK_SD_CARD_WRITE_MIN] = (u8) ((u64) sd_MaxAccessedBlockWriteMin_u32 * (u64) 100 / (u64) Blockcount_u32) + 1;   
    output[OUTPUT_CMD_RESULT_OFFSET + DEVICE_HIGH_WATERMARK_SD_CARD_WRITE_MAX] = (u8) ((u64) sd_MaxAccessedBlockWriteMax_u32 * (u64) 100 / (u64) Blockcount_u32) - 1;  
    output[OUTPUT_CMD_RESULT_OFFSET + DEVICE_HIGH_WATERMARK_SD_CARD_READ_MIN] = (u8) ((u64) sd_MaxAccessedBlockReadMin_u32 * (u64) 100 / (u64) Blockcount_u32) + 1; 
    output[OUTPUT_CMD_RESULT_OFFSET + DEVICE_HIGH_WATERMARK_SD_CARD_READ_MAX] = (u8) ((u64) sd_MaxAccessedBlockReadMax_u32 * (u64) 100 / (u64) Blockcount_u32) - 1; 
```

Therefore the 50/49% can appear. Probably reading was accessed on the SD card with an address> 50% and writing <= 50%.
Then the rounding process is min = 50% and max = 49%. The easiest way to correct the thing on the app is to prevent the Min value from being larger than the Max value.

szszszsz, 2017-03-21:
Moved to [Nitrokey App ](https://github.com/Nitrokey/nitrokey-app/)issues.

closed 2017-03-21 by szszszsz