Nitrokey/nitrokey-storage-firmware #33: Device is connected after 20 minutes on latest Windows 10

szszszsz, 2017-04-10:
Windows 10 Creators Update
Details: https://support.nitrokey.com/t/nitrokey-storage-extrem-slow-after-windows-10-creators-update/431/4
Quote:
> After upgrading my computer to Windows 10 Creators Update (version 1703) mounting Nitrokey Storage is extremly slow.
> It tooks ~15-20 minutes after insserting the stick to recognise the unencrypted volume and also to recognise the stick in the nitro key app.
> After this time i can unlock the encrypted volume....and have to wait again to mount the unlocked volume.

silvioboehme, 2017-04-10:
I checked this again on a Windows 10 Anniversary (1607) and stick is working fine.
Also checked this on a complete different machine with Win10 Creators Update (1703) with same issue as above.
So I think it's easy reproducible!

Just put a NitroKey Storage in a computer with a current Win 10.

szszszsz, 2017-05-30:
I have retested this issue by the occasion of other check. Details:
Storage v0.45
Windows W10 1703 15063.338 - device connection is detected by the OS but it is not powering up the device.
Here is a Windows error message from Event Viewer:
```
Device USB\VID_20A0&PID_4109\0000000000000 had a problem starting.

Driver Name: usb.inf
Class Guid: {36fc9e60-c465-11cf-8056-444553540000}
Service: usbccgp
Lower Filters: 
Upper Filters: 
Problem: 0xA
Problem Status: 0xC000000D
```

silvioboehme, 2017-05-30:
I don't see such a error message.
Does it appear directly after you plug in the stick or after "some" time?

szszszsz, 2017-05-30:
This one is displayed in system's log. You can find it in `Device Manager` -> device's `Properties` -> `Event Log`

silvioboehme, 2017-05-30:
On which device? Smartcard Reader?

szszszsz, 2017-05-30:
[image]
In my case it is `USB Composite Device`.

silvioboehme, 2017-05-30:
Nope...nothing. Everything is fine....except Nitrokey isn't working.

szszszsz, 2017-05-30:
Just to note, what build exactly do you have? You can find it in `System` window.
One more idea: have you tried to enable `Legacy USB` (or similar) in BIOS/UEFI?

silvioboehme, 2017-05-30:
1703 15063.296

silvioboehme, 2017-05-30:
Windows W10 1703 15063.338 doesn't exists???

szszszsz, 2017-05-30:
Indeed - typo, I meant .332, sorry.

silvioboehme, 2017-05-30:
Btw..this is the first reaction after Microsoft relased the Creators Update!
... 2 month ago
... which is sad!

Also the bug title is a little bit "missleading". It is just unusable on a current windows 10!
After the stick is connected after about 10-20 minutes (connected means visible as a drive) it still isn't usable.
The drive letter appears twice, and the Nitro APP doesn't find the stick. (see another forum post!)
Also...if i copy a file to the unencrypted drive (on one drive letter), the file doesn't appear on the "other" drive....which initialy showed the same files.
 
Sadly...I have payed 99€ for nothing!

juliomaranhao, 2017-06-10:
I have also this issue. Summarising my case: 

- UEFI only (Legacy Mode disabled) and Secure Boot disabled;
- Win 10 Pro, version 1703, build 15063.332, AKA "Creators Update", AKA "Current Win 10", AKA "I stayed with Anniversary as long as I could";
- Windows is in "Test Mode" to allow me to install unsigned drivers (VirtualSmartCard). Just for your info;
- Nitrokey Storage 8 GB, almost clean (never used), firmware version 0.45, App version 1.1;
- From image below: 2 virtual smartcard readers, Nitrokey and a gemalto (USB) smartcard reader. The command `opensc-tool -l` shows 3 smart cards but not Nitrokey. NitroApp can detect the Nitrokey.
[image]
```
C:\Program Files\OpenSC Project\OpenSC\tools>opensc-tool.exe -l
# Detected readers (pcsc)
Nr.  Card  Features  Name
0    No              Gemalto USB Smart Card Reader 0
1    Yes             MySmartLogon Virtual Smart Card Reader 0
2    Yes             Virtual Smart Card Architecture Virtual PCD 0
```
[image]
"Não há drivers instalados para este dispositivo" = "No drivers installed for this device"
[image]

About the sequence:
- Nitrokey pluged;
- Device Manager lists "Nitrokey Storage" smartcard reader (above pics);
- Nitrokey App does not detect the Nitrokey and no drivers in windows explorer;
- Many minutes later:  2 drives appear (same label and contents) with 2 GB FAT32 each and NitroApp detects Nitrokey. Disc Manager also shows 2 new discs;
- `opensc-tool -l` still cannot show Nitrokey.
- And no errors in Event Viewer but I got this Information Event (translated) `Device USB\VID_20A0&PID_4109&MI_01\7&57a3ad1&0&0001 needs adicional installation.`
 
I have some time and knowledge to help with this bug ("windows advanced user") and I don't need Nitrokey now (I am evaluating it). But I agree with silvioboehme. The Nitrokey project needs help to reach a "product" status. I see this in the documentation. It's at least sparse. Maybe even scarse. And a Forum or FAQ is not the same as a Manual.

Anyway, I am here to help.

juliomaranhao, 2017-06-12:
Me again. I decided to test my Nitrokey Storage 8GB in a Win7 Pro.

Results from 1st insertion:

[image]

Results from "remove and reinsert" AKA 2nd insertion:

[image]

Then, results from opensc-tool:
```
C:\Program Files\OpenSC Project\OpenSC\tools>opensc-tool -l
# Detected readers (pcsc)
Nr.  Card  Features  Name
0    Yes             Nitrokey Nitrokey Storage 0

C:\Program Files\OpenSC Project\OpenSC\tools>opensc-tool -n
Using reader with a card: Nitrokey Nitrokey Storage 0
CryptoStick v1.2 (OpenPGP v2.0)
```
Nitrokey App is working (version 1.1, firmware 0.45) and Explorer shows only 1 unencrypted drive (2 GB FAT32).

From this and my previous post:
- My Nitrokey Storage 8GB is NOT broken (works in Win7);
- In Win7 it takes 4 files (3 sys and 1 dll) in driver's list. All Microsoft files;
- In Win10 it takes only 1 file (1 sys) in driver's list. A Microsoft file;
- In Win10: "No drivers installed for this device";
- In Win7: "This device is working correctly" (Translated).

It's related to device detection and drivers installation. Even in Win7 I need to remove and reinsert for it to work (only 1st time ever, of course). This does not seem right as it's not a common behavior for other types of devices. Anyway it's a clue.

juliomaranhao, 2017-06-19:
Two posts above I said my NK Storage was (almost) never used. I forgot I did a firmware upgrade from 0.4x (bought in december 2016) to 0.45. Now I only remember the PINs. I also tested the defaults. No luck with the firmware password.

So I am quitting as my NK is stuck with 0.45.

jans23, 2017-06-19:
This issue should be solved by firmware [0.46](https://github.com/Nitrokey/nitrokey-storage-firmware/releases/tag/V0.46.0). Please [update](https://www.nitrokey.com/en/doc/firmware-update-storage) your firmware and try again.

closed 2017-06-19 by jans23