Nitrokey/nitrokey-storage-firmware #36: Some commands combinations crashes the device

szszszsz, 2017-05-23:
v0.45
App v1.1
Windows 7

Some device commands sequential combinations crashes the device or makes the device `BUSY` (as it is reporting through HID interface). 
Example sequence for the latter case are `GET_STATUS` (Pro status command) and `GET_DEVICE_STATUS` (Storage status command) invoked one after another - in some cases it makes the device `BUSY` for 20 seconds, but later it functions properly.
Another case is getting `Password Safe` status just after unlocking encrypted volume (more details needed).

Log:
[windows7-devicebusy-log.txt](https://github.com/Nitrokey/nitrokey-storage-firmware/files/1022732/windows7-devicebusy-log.txt)

szszszsz, 2017-05-27:
Similar issue: #31

szszszsz, 2017-05-29:
Issue has occurred today on modified v0.45, but ceased showing up after factory reset and re-initializing AES keys.

jans23, 2017-06-14:
@szszszsz Could you provide a reproducable example, custom binary (for Windows) or simple script to reproduce this issue?

szszszsz, 2017-06-14:
This issue was occurring sometimes with [App v1.1](https://github.com/Nitrokey/nitrokey-app/releases/tag/v1.1). Possibly on inserting the device after the Application was run. Could you reinsert the device a couple (10-20) times and see in App logs would device respond with `BUSY` status after `GET_DEVICE_STATUS`? I expect it will reply with OK immediately on each request.

Hopefully I will prepare some binary tomorrow which would always reproduce this behaviour.

CC @rudbo

jans23, 2017-06-15:
When the App is started and I connect a NK Storage and click on the App's icon fast enough, I often get a "Smart card or storage not available". Once displayed, this status can only be avoided by reconnecting the NK Storage. Also restarting the App doesn't help. Perhaps this is the same issue?

szszszsz, 2017-06-16:
@jans23 This one could be connected - I suspect both issues are caused by some faults in communication with smartcard.
Application already repeats request for the `GET_DEVICE_STATUS` after it receives `not available` status of the smartcard, but gives up ultimately.

szszszsz, 2017-08-16:
Issue is not occurring on v0.47. Closing.

closed 2017-08-16 by szszszsz