Nitrokey/nitrokey-storage-firmware #39: resize unencrypted storage partition?

stgarf, 2017-06-10:
>How large is the storage capacity?

>Nitrokey Pro, Nitrokey Start, Nitrokey HSM and Nitrokey U2F don't contain storage capability for ordinary data (it can only store cryptographic keys and certificates).
>Nitrokey Storage can store and encrypt 8, 32, or 64 GB of data (depending on particular model).

Upon receiving the device this doesn't appear to be the case. I have a 2GB unencrypted partition and after initializing the device there's another ~5.5GB encrypted device that I'm able to format to whatever filesystem I wish. My question is, how do I reclaim some of that 2GB and shrink it. Or in fact, how can I expand it -- for instance, to install Arch+i3+GRUB etc (might need a bit more than 2GB).

szszszsz, 2017-06-14:
Hi!
I have seen in one of the forks someone has changed the unencrypted/encrypted ratio:
https://github.com/mupuf/nitrokey-storage-firmware/commit/14e6a786f6875fb783b3a3bff24abdaf6a1e901b
I do not know how is it supported from official side.

szszszsz, 2017-06-23:
Let me know if I understand this correctly: what you would like to do is to change unencrypted/encrypted ratio of the storage without changing/replacing the firmware, is that right?

stgarf, 2017-06-24:
Yes but if theres diff firmware thats needed to do that that would work but
flashing might be prohibitive for some people.

szszszsz, 2017-06-24:
I see. I have updated the title then to better express that. Indeed it might be a nice function, I do not know though how is it with its feasibility.

Peacekeeper2000, 2018-04-11:
I would support this too !

closed 2024-01-06 by stgarf