Nitrokey/nitrokey-storage-firmware #41: Device crash/ejection on unencrypted volume unlock / WUDFHost.exe

szszszsz, 2017-06-20:
#### Description
Issue frequency: once
During the tests suddenly the device stopped responding. Additionally:

- WUDFHost.exe started to take 100% of one of the cores, 
- Application has shown tray message about the device being disconnected, unencrypted
volume vanished, 
- `usbview` shows device's general failure, 

#### Environment
Windows 8.1
USB 2.0
App v1.1
Storage v0.46
Device state: after libnitrokey tests (done on Ubuntu a minute before)

#### Scenario
1. Check if unencrypted volume shows up and is accessible
2. Check PWS/OTP functionality
3. Try to unlock encrypted volume

#### Logs
Excerpt from system log below (looks like issue was caused by CCID, but possibly linked with device ejection by the OS)
```
- <Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
- <System>
  <Provider Name="WudfUsbccidDriver"
Guid="{22C370A7-A3DB-4390-ADE5-3A1ACCF4B5D5}" />
  <EventID>7</EventID>
  <Version>0</Version>
  <Level>2</Level>
  <Task>1</Task>
  <Opcode>10</Opcode>
  <Keywords>0x8000000000000000</Keywords>
  <TimeCreated SystemTime="2017-06-15T11:10:47.423497100Z" />
  <EventRecordID>1901578</EventRecordID>
  <Correlation />
  <Execution ProcessID="5932" ThreadID="7520" />
  <Channel>System</Channel>
  <Computer>sz1070</Computer>
  <Security UserID="S-1-5-19" />
  </System>
- <EventData>
  <Data Name="Name">ReaderCompletionUnknownMsgType</Data>
  <Data Name="Value">0x0</Data>
  </EventData>
  </Event>

Device responded with an error status.
Status: ReaderCompletionUnknownMsgType
```

jans23, 2017-08-23:
@szszszsz Is this issue still valid with firmware 0.48?

szszszsz, 2017-08-24:
Issue has not occurred on Windows 8.1 with firmware v0.48. Due to `once` frequency it will be left open to retest on v0.49 and then closed, if it will not show up.

szszszsz, 2018-02-16:
Issue has not occurred since then. Closing.

closed 2018-02-16 by szszszsz