Nitrokey/nitrokey-storage-firmware #42: Rejected IOCTL on Windows 8

szszszsz, 2017-06-20:
#### Description
While reading event log on Windows 8 I have encountered on following error event:
```
Smart Card Reader 'Nitrokey Nitrokey Storage 0' rejected IOCTL 0x313520:
Incorrect function.  If this error persists, your smart card or reader
may not be functioning correctly.
```
#### Environment
Windows 8.1
USB 2.0
App v1.1
Storage v0.45.5
#### Scenario
Insert the stick and check event log.

szszszsz, 2017-08-04:
Occurs on v0.46.5

jans23, 2017-08-23:
@szszszsz Is this issue still valid with firmware 0.48?

szszszsz, 2017-08-24:
Just tested. Yes, it still occurs on Windows 8.1, firmware v0.48.

rudbo, 2017-08-25:
The error message of the event log of WIN7 tells more:

Fehler bei einem Vorgang (0x6, 0x0, 0x0, 0x0).
ScCardPowerDown: IccPowerOff failed.
HResult: {Vorgang fehlgeschlagen}
Der Vorgang konnte nicht durchgeführt werden.

The error is only a information that the smart card isn't in the power off state.

This is normal, because the firmware accesses the smartcard by itself and via APP command. In this cause the CCID interface powerdown the  smart card but by the internal usage the smartcard is powered on again, so that the power state is on.

closed 2017-08-25 by rudbo