Nitrokey/nitrokey-storage-firmware #45: Constant auto mounting the device under macOS 10.12.6

Peacekeeper2000, 2017-08-31:
FW v0.46+ don't allow to unmount/eject the device through the finder. When you try to do so, the device disappears as expected, but show up again within a few seconds. So unplug of the device could destroy the device as it is still attached.

Workaround: sudo diskutier unmountDisk /dev/diskx where disk is the disk representing NK Storage.
FW v0.45 has not that behavior.

szszszsz, 2018-03-06:
Reported as well in https://support.nitrokey.com/t/lost-encrypted-volume/969

oderwat, 2018-03-29:
Same happens here which lead me to this issue. I understand that after ejecting the volume is flashed it still bugs me a lot. Is this going to be fixed in a later firmware? We got a new release in January 2018 and it still happens afais.

szszszsz, 2018-03-30:
Hi @oderwat !
That is a great tip! I wonder could I incorporate that into Nitrokey App's Unlock/Lock procedure as a form of workaround. 
As for the issue, I hope this would be fixed as well. I do not know any dates though. It will probably not be done in v0.51 looking at the [milestone page](https://github.com/Nitrokey/nitrokey-storage-firmware/milestone/1).

oderwat, 2018-03-30:
Well. I tried around with using hidden volumes and got multiple data loss because of the auto mounting. You simply can't switch between the volume because you always end up with OSX "thinking" that the previous volume is still existing. You also can't reliable hinder OSX to write to volumes. When using "deavtivate" you still need to "eject" for the old volume to vanish and get the hidden volume. Is anybody is using this with relevant data in a reliable way?

oderwat, 2018-03-31:
I made a proposal here https://github.com/Nitrokey/nitrokey-app/issues/337#issuecomment-377646808 about what imho is a better way to implement multiple volumes.

szszszsz, 2019-01-14:
From: https://github.com/Nitrokey/nitrokey-app/issues/337#issuecomment-377646808
> I actually think that the hardware should actively be removed and inserted again between the volume changes, just like a real USB stick!

Perhaps the quickest way to achieve that is to disable the Windows 10 SCSI compatibility flag for the macOS users, which would restore the expected behaviour (as seen in v0.45 and lower). 
- [ ] To check, whether it could be user-configurable (for cross-platform usage).

szszszsz, 2019-01-14:
Related: https://github.com/Nitrokey/nitrokey-storage-firmware/issues/75

Peacekeeper2000, 2019-01-14:
Is there not a merge possibilities for both issues ?  So you mean to disable Win 10 SCSI in the firmware and compile it again ?

szszszsz, 2019-01-14:
@Peacekeeper2000 This is only an idea - I have not tested this yet, please be patient.
Let me keep this issue open until we will not solve this.

jans23, 2020-04-03:
> Perhaps the quickest way to achieve that is to disable the Windows 10 SCSI compatibility flag for the macOS users, which would restore the expected behaviour (as seen in v0.45 and lower).

@szszszsz Please describe this change. Also, could you refer to the appropriate commit?

szszszsz, 2020-04-03:
@jans23  That's right, https://github.com/Nitrokey/nitrokey-storage-firmware/commit/2c123f60ff9feb506a1670772cc15e8db30b024. I have just checked and there is no other commit since v0.45, which would be related to SCSI.