Nitrokey/nitrokey-storage-firmware #46: Improve performance

jans23, 2017-09-04:
Firmware 0.48 provides about >3 MB/s write performance only.

szszszsz, 2017-09-20:
On Windows 8 I observe 4.8 MB/s on average while copying 175 MB file to hidden volume. There was a speed drop for a second under 3 MB/s, but nothing more. 
I would like to reproduce it on next tests. On what OS have you observed the issue? What was the structure of copied data (big: >100MB, medium:>1 MB or small files)?

jans23, 2017-09-20:
On Linux I observed about 3.5 MB/s when copying a ca. 500 MB file to the encrypted volume with the following commands:

`time cp a-large-movie.avi /media/jan/Nitrokey/; sync`
`dd if=/dev/zero of=/media/jan/Nitrokey/testfile bs=512M count=1 oflag=direct`