Nitrokey/nitrokey-storage-firmware #48: Set unencrypted volume writable with Admin PIN?

jans23, 2017-11-03:
This idea needs to be confirmed!

Once the Nitrokey App is preinstalled on the unencrypted volume, it should be set read-only by default and only allow the Admin PIN to change it.

szszszsz, 2017-11-06:
Currently User PIN is needed to change the state of the unencrypted volume and AFAIR it is in read-write by default.

szszszsz, 2017-12-20:
Confirmed new command is working with Admin PIN with test firmware.

jans23, 2018-01-28:
How can this be implement without letting users (of old NK App) run into backward-compatibility issues such as decreasing PIN counter? Perhaps remove the old command and add a new command instead of changing the existing one? Any other idea?

szszszsz, 2018-01-29:
In v0.49 it was implemented in such way - command requiring User PIN always replies with 'wrong password' now (without decreasing the retry counter). The one requiring Admin PIN has new ID thus needing from host information about the protocol change and newer version of client applications. 
To have this decoupled, one way is to make the device an API server, similarly to web technologies but via USB (that is its commands and parameters could be listed and described). Perhaps part of it would be feasible, e.g. querying only for PIN type for given command.

szszszsz, 2018-11-28:
I think this is fixed. Please close on confirmation.

closed 2018-11-30 by jans23