Nitrokey/nitrokey-storage-firmware #52: Present unlocked state by blinking LEDs

szszszsz, 2017-11-07:
Device could present when it is in unlocked state (PWS or Encrypted/Hidden volumes are available) through blinking with one of the LEDs. This would remind user about locking it after use.

jans23, 2017-11-07:
I think that would be too disturbing when used on a laptop.

closed 2017-11-07 by jans23

szszszsz, 2017-11-08:
I think you are right. And since both LEDs are already assigned to separate functions (CCID and HID) it might be confusing too. I was inspired by similar product feature (I think it was GNUK).
Perhaps an auto-locking function would be better instead (eg. lock after 10 minutes from last use of encrypted volume / CCID / HID).

jans23, 2017-11-09:
Auto locking could implicate unforeseen issues too. For example if somebody uses portable apps continuously, it would be bad if the storage media disappears.
-- 
Sent from a mobile device.