Nitrokey/nitrokey-storage-firmware #53: Insert static string on special-key double press

szszszsz, 2017-11-08:
Allow to configure and insert a password string on a special-key double press, similarly to current HOTP functionality.

szszszsz, 2018-03-21:
Perhaps using smartcard's DOs 0101-0104 (for private use) would do ([OpenPGP v2.1](https://gnupg.org/ftp/specs/OpenPGP-smart-card-application-2.1.pdf), page 16 top).

jans23, 2018-03-23:
I understood this ticket is about extending the current feature "double press special key to release HOTP" to be used with static passwords (instead or in addition to HOTP). But now I don't understand how this is related to DOs of the OpenPGP Card.

szszszsz, 2018-03-23:
By the occasion of visiting OpenPGP v2.1 specification and reading about mentioned DOs I had a thought that we could leverage this storage for static passwords use (in addition to regular Password Safe). I did not make any deep analysis, just an idea. Perhaps this would conflict with user usage.