Nitrokey/nitrokey-storage-firmware #55: Unlock encrypted volume in read-only mode

szszszsz, 2017-11-30:
Allow user to unlock encrypted volume in read-only mode. 
Perhaps extend this to hidden volume as well.

See request from https://github.com/Nitrokey/nitrokey-app/issues/296

ljay79, 2023-12-20:
Im new to Nitrokey, using Nitrokey Storage 2. Beside using it as enrypted un-frequently used backup storage stick i would like to enhance daily usage.
Using the encrypted volume with read-only as default would be awesome for daily use as a data storage with daily used sensitive data. But to prevent accidental writing/changes or even secure for malware, it would be very good to have it open in read-only be default or at all.

So far, i could not find this feature available. Is this coming at all? To storage 2?

daringer, 2023-12-22:
hey @ljay79, for the Nitrokey Storage 2 no new features are developed anymore, this means that most likely the won't be such a feature for this device.