Nitrokey/nitrokey-storage-firmware #58: Double mounting unencrypted volume macOS 10.13

szszszsz, 2018-03-06:
Devices unencrypted volume is presented two times in the OS and the encrypted volume is not shown at all.
Details: https://support.nitrokey.com/t/lost-encrypted-volume/969
OS: macOS 10.13
Device: Storage v0.50

jans23, 2018-03-08:
@szszszsz @alex-nitrokey can you get instructions how to reproduce the issue, please?

szszszsz, 2018-03-15:
User reports no specific reproduction scenario, ordinary use only.

szszszsz, 2018-04-04:
Perhaps connected with #60

szszszsz, 2018-05-28:
Maybe we could distribute second firmware with Windows 10 compatibility reverted for the users (https://github.com/Nitrokey/nitrokey-storage-firmware/commit/2c123f60ff9feb506a1670772cc15e8db30b0241), which prefer auto-mounting to be disabled instead.

szszszsz, 2018-10-25:
Same confirmed on Ubuntu Mate 18.04, source: [how-to-use-nitrokey-app-in-ubuntu-w-gnome-3-26](https://support.nitrokey.com/t/how-to-use-nitrokey-app-in-ubuntu-w-gnome-3-26/1386/7?u=szszszsz)