Nitrokey/nitrokey-storage-firmware #60: Make Encrypted Volume lock more noticeable for the OS

szszszsz, 2018-03-21:
Make Encrypted Volume lock/removal more noticeable for the OS to make it invalidate the file system cache. 

Currently in specific scenario system's cache allows to list files on EV (but not their contents, unless processed earlier; confirmed) after the smartcards PIN reset (via CCID) and unlocking with the default PIN, when the EV's file system was not unmounted properly before that operation.

This is not a security issue of the device - cause is in the OS file system cache, which is not recognizing that the EV block device has been 'cleared' (and EV's partition table stopped to exist). This issue is opened in a hope that device could better inform the OS about EV change, so it would invalidate its cache.

Environment: Ubuntu 18.04 (reported originally on Linux Mint), Storage v0.50
Details / source: [forum/verschlusselungspasswort-geknackt-pw-cracked](https://support.nitrokey.com/t/verschlusselungspasswort-geknackt-pw-cracked/907/8)

Scenario:
Preparation: Run Nitrokey App. A second later insert Storage device.
1. Change User PIN to "cracked" (from "123456")
2. Unlock EV with default User PIN (fails)
3. Unlock EV with changed User PIN (works)
	- test EV content (mount EV)
4. Lock EV only (NOT 'lock device'), leave EV mounted
5. Run `factory-reset` via `gpg2 --card-edit`
6. Run step 3 (fails)
7. Run step 2 (works)
	- test EV content, use existing EV mount (done in step 3)

alex-nitrokey, 2018-10-29:
This issue turned out to be another problem not related to system cache imho, it was the missing AES key issue solved by [v0.51](https://github.com/Nitrokey/nitrokey-storage-firmware/releases/tag/V0.51).

Or am I wrong?

szszszsz, 2018-10-29:
That is correct, but I have not found time yet to update this issue. Could you make a quick summary (what was the cause, how it was handled, and when firmware was released) and close it please?

alex-nitrokey, 2018-10-29:
Doing a factory-reset via GnuPG does reset the internal OpenPGP Card and thus provoked a AES DO which only contains zeros. Doing a factory-reset via the Nitrokey App always creates a new AES key, so this did only happen if GnuPG's factory-reset and nothing else was used.

The Nitrokey Storage did not reject a zeroes AES key for encryption. Therefore, a user may had used the encrypted volume with the most simple AES key.

This is prevented since [v0.51](https://github.com/Nitrokey/nitrokey-storage-firmware/releases/tag/V0.51) from June 14. The Nitrokey Storage refuses to decrypt the encrypted volume if the AES key is zero.

closed 2018-10-29 by alex-nitrokey