Nitrokey/nitrokey-storage-firmware #62: ECC support for nitrokey storage

FrostKnight, 2018-04-09:
Simply put, I would like to see at some point, nitrokey storage to be protected by ecc, although it would need to be curve25519.

When I put my files into nitrokey storage, I would like them to be protected by that curve or a better one if it even exists...
curve25519 though seems to be the best choice at the moment.
I expect this won't happen immediately, but for the future that is what I desire.
I am also wondering how many years such a feature would take. Feel free to reply whenever...

jans23, 2018-04-09:
ECC-support will be solved by the upcoming Nitrokey Storage 2, but not for Curve25519.

FrostKnight, 2018-04-10:
On 04/09/18 13:49, jans23 wrote:
> ECC-support will be solved by the upcoming Nitrokey Storage 2, but not for Curve25519.
>
Hmm curious, will it be a curve as secure as Curve25519? or more secure?

jans23, 2018-04-10:
It will be Brainpool and NIST.

jans23, 2018-06-26:
The currently shipped Nitrokey Storage supports ECC (Brainpool and NIST).

closed 2018-06-26 by jans23