Nitrokey/nitrokey-storage-firmware #64: Use memset_s

jans23, 2018-05-23:
The code contains several occurrence of memset. memset could be optimized away by the compiler which is why memset_s should be used instead, for cases where a key or password or other confidential data has to be overwritten im RAM.

NKelias, 2019-01-24:
`memset_s` seems to be a feature of C++11 and is not available on the AVR32 platform.

Alternatively, we could add a safe erase function as proposed here:
https://www.cryptologie.net/article/419/zeroing-memory-compiler-optimizations-and-memset_s/

szszszsz, 2019-01-25:
Indeed, `volatile` should do the job. It is interesting to see double `volatile` in the [libsodium implementation]. 

[libsodium implementation]: https://github.com/jedisct1/libsodium/blob/be58b2e6664389d9c7993b55291402934b43b3ca/src/libsodium/sodium/utils.c#L78:L101

PR #85 (2019-06-13): v0.54 updated OTP handling
This reworks the OTP handling to resemble that of the Nitrokey Pro.

### Functional changes:
- OTP secret extended to 40 Bytes
- Authorization changed from pre-authenticated CRC to temporary password in HID report
- OTP counter transferred as 64bit unsigned Integer instead of C-String
- OTP counter is retained when editing slots
- Temporary passwords cleared through `lock_device` operation
- HOTP verification functionality added

### Further changes:
- OTP handlng now uses struct format for message parsing and passing data around
- Replace optimizable `memset` with n [... cut]
closed 2019-06-13 by szszszsz