Nitrokey/nitrokey-storage-firmware #76: GET_STATUS command reports wrong smart card serial number

robinkrahl, 2018-12-27:
The serial numbers reported by `gpg --card-status` and by the `GET_STATUS` command diverge for my Nitrokey Storage:

```
$ gpg --card-status
[…]
Serial number ....: 0000636F
[…]
$ ./sn-test
[…]
[Thu Dec 27 12:45:03 2018][DEBUG_L1]	=> GET_DEVICE_STATUS
..
[Thu Dec 27 12:45:04 2018][DEBUG_L1]	<= GET_DEVICE_STATUS 0 1
[Thu Dec 27 12:45:04 2018][DEBUG]	Incoming HID packet:
[Thu Dec 27 12:45:04 2018][DEBUG]	Device status:	0 OK
[…]
 ActiveSmartCardID_u32:	327680
[…]

Serial number: 00050000
```

`sn-test.c`:
```c
#include <stdio.h>
#include <libnitrokey/NK_C_API.h>

int main(void)
{
	NK_set_debug(true);
	int err = NK_login_auto();
	if (err == 0) {
		fprintf(stderr, "Could not connect to Nitrokey.\n");
		return 1;
	}
	char *sn = NK_device_serial_number();
	printf("Serial number: %s\n", sn);
	return 0;
}
```

This seems to be a bug – or am I missing something?

szszszsz, 2018-12-27:
Hi!
It looks like a bug indeed. libnitrokey runs proper status calls for each device ([NitrokeyManager.cc#L361-L376](https://github.com/Nitrokey/libnitrokey/blob/master/NitrokeyManager.cc#L361-L376)), so it might be this is a Storage's firmware side issue.

robinkrahl, 2020-01-05:
I just noticed that this is the `GET_DEVICE_STATUS` command, not `GET_STATUS`.

robinkrahl, 2020-01-06:
As far as I see, the problem must be in one of the following functions:
1. [`GetSmartCardStatus`][] in `USER_INTERFACE/html_io.c`
2. [`LA_OpenPGP_V20_GetAID`][] in `CCID/LOCAL_ACCESS/OpenPGP_V20.c`

But as I am not familiar with the OpenPGP card interface, I can’t track it down any further.

[`GetSmartCardStatus`]: https://github.com/Nitrokey/nitrokey-storage-firmware/blob/7d2e9ce4010497a4230dc6dd748458aa502788db/src/USER_INTERFACE/html_io.c#L942
[`LA_OpenPGP_V20_GetAID`]: https://github.com/Nitrokey/nitrokey-storage-firmware/blob/9f1c19f74e169ebc055af78e5bbd52a39a85da84/src/CCID/LOCAL_ACCESS/OpenPGP_V20.c#L1108