Nitrokey/nitrokey-storage-firmware #78: Distribute via “Linux Vendor Firmware Service” (LVFS)

rugk, 2019-01-13:
As I elaborated [in this forum thread](https://support.nitrokey.com/t/updates-on-with-linux-vendor-firmware-service-lvfs/1582) it would be great to have this distributed with [LVFS](https://fwupd.org/), which enables Linux users across many distros to do firmware updates of your device in their system's _graphical tool_ rather than on the command line.

AFAIK they also offer a way to show a message before the upgrade, where you can hint users to set the Nitrokey into "update mode" first.
So I see no technical reasons preventing easy firmware upgrades on Linux via this great service! :smiley:

szszszsz, 2019-01-14:
Confirmed. Current firmware on fwupd is v0.50, while the latest is v0.53.

rugk, 2019-01-14:
It's also marked as testing, while the description says it is stable. See also my forum thread.

rugk, 2019-01-14:
@szszszsz BTW, why did you add the "invalid" label?

szszszsz, 2019-01-14:
Last time I remember it was meant to be stable, I wonder why it stayed testing. Will check.

`Invalid` label is to mark issues, which are not actions from the source code POV (e.g. not bugs, features, compatibility changes etc). This is a custom, which is used in reporting tools. It sounds a bit negatively though; perhaps `task` label would be sufficient to show that instead. Will add description to this label nevertheless.

alex-nitrokey, 2019-01-14:
@szszszsz let me know as soon as you updated the files and settings upstream. I can test the procedure if you like.

szszszsz, 2019-01-23:
Main issue with the `fwupd`-based updating is fixed! Two tasks left:
- to test updates from the older firmwares to current (I have tested v0.53->v0.53);
- to fix the firmware version reading plugin, which shows `0.0` at the moment, and might confuse users. It will be shipped with the next release of the `fwupd`, so it must be done before that.

rugk, 2019-01-23:
Also about the "verify upgrade" batch, that would also be a useful feature, I guess.☺

szszszsz, 2019-01-30:
Right. We have talked about that, and automatic verification should be feasible to do.

Regarding the `fwupd`-based update, plugin for it is fixed now (https://github.com/hughsie/fwupd/pull/961). Waiting until its next release (should be next month, first half).

hughsie, 2019-01-30:
Release should be on the 1st. You can depend on fwupd 1.2.4 in the interim if that helps.

4jNsY6fCVqZv, 2019-10-05:
Hi, I find this issue very worth supporting, thank you for working on it!

However, I have a practical question: What does it look like if I have a Nitrokey storage and want to update it using fwupd?

* The stick is never permanently plugged in and therefore part of my system. How does my system recognize that an update is available?
* I also wonder if the update means that all information and keys on the stick will be overwritten by the update.
* Can I use fwupd to scrape my key with an update? Is there a risk?

rugk, 2019-10-05:
Also there is already [v0.54 released](https://github.com/Nitrokey/nitrokey-storage-firmware/releases/tag/V0.54), but [fwupd/LVFS is still at v0.5.3](https://fwupd.org/lvfs/device/com.nitrokey.storage.firmware). So can you please also update it on LVFS?

4jNsY6fCVqZv, 2019-10-05:
When I look at https://fwupd.org/lvfs/device/com.nitrokey.storage.firmware, two questions also arise for me:

1. Is it possible that the Nitrokey updates are automatically imported from the GitHub repository into LVFS so that the packages are always up to date?

2 The overview shows that the Nitrokey package does not meet two security requirements:
a) Update is not cryptographically signed
b) Firmware cannot be verified after flashing 
Would it make sense and be possible for you to fulfill them?

rugk, 2019-10-05:
2. Yes, this [has been discussed in the forum already](https://support.nitrokey.com/t/updates-on-with-linux-vendor-firmware-service-lvfs/1582/3?u=rugk). At least the automatic test [should be possible, as it has been confirmed before](https://github.com/Nitrokey/nitrokey-storage-firmware/issues/78#issuecomment-458935871).

hughsie, 2019-10-05:
If it helps, the LVFS has an account type for automated "robot" uploads. It's how a few of the big OEMs manage all the uploads to the LVFS.

4jNsY6fCVqZv, 2019-10-05:
@hughsie Could you please share a link where the setup of such a feature is documented?

hughsie, 2019-10-05:
It's not documented, it's the kind of thing I help the vendor with as required. Obviously there are a few authentication-type things to set up.

4jNsY6fCVqZv, 2019-10-05:
Thank you, it would be wonderful if you could support the developers of Nitrokey!
What do you say, @szszszsz?

rugk, 2019-10-06:
…also would possibly be a good idea to "standardize" it (?) and document it… (undocumented features are usually not good)