Nitrokey/nitrokey-storage-firmware #83: Unable to export firmware properly

muellermartin, 2019-01-18:
- Application version: v1.3.2
- Operating system: macOS 10.13.6 (64-bit)
- Device model and firmware version: Storage v0.53
- Connected USB port type: 3.0
- Issue occurrency: sometimes

### Expected behaviour

Firmware can be exported into a file and this file can be successfully matched against the original release of the same version.

### Current behaviour

Mostly the exported `firmware.bin` file is not listed in the file system of the USB storage although the notification says it was successfully exported.

When the Nitrokey Storage is remounted several times and/or the firmware export is repeated (not sure what helps there), the file appears, but is somewhat faulty. For example, if the file is copied to disk, the Finder shows following error message (German):

> Der Finder konnte diesen Vorgang nicht abschließen, da einige Daten in „firmware.bin“ nicht gelesen oder geschrieben werden konnten.
> (Fehler: -36)

Nevertheless, the file gets copied – at least it seems so (there is a copy and it contains data). Because when using `nkcompare.sh` from [Nitrokey/nitrokey-storage-firmware](https://github.com/Nitrokey/nitrokey-storage-firmware) `/tools` to compare the exported binary with the original one from GitHub, the binaries do not match ("Firmware binaries mismatch"). This is likely because the exported `firmware.bin` contains very few data (only exact 4096 bytes!) and the rest is filled with zeros (see attached `firmware.bin`).

When using `cp` instead of Finder for coyping the `firmware.bin` file from the USB storage to the disk, following error is shown:

```
$ cp /Volumes/Nitrokey/firmware.bin .
cp: /Volumes/Nitrokey/firmware.bin: Input/output error
```

And the resulting copy is an empty file (0 bytes).

Deleting the file from the Nitrokey storage works, though.

Directly reading the file from the USB storage (e.g. `xxd /Volumes/Nitrokey/firmware.bin | less` also gives an `Input/output error`).

### Steps for reproduction

#### Preconditions
Nothing special.

#### Steps
1. Insert Nitrokey Storage
2. Start Nitrokey App
3. Select tray icon > Configure > Export firmware to file and wait until notification signals completed export.
4. Repeat 1-3 several times until `firmware.bin` is visible in the filesystem of the USB storage.
5. Try to copy or read the `firmware.bin` (e.g. for comparison with original firmware)

### Logs

Logs attached: 
- [debug-file-no-firmware.txt](https://github.com/Nitrokey/nitrokey-app/files/2752694/debug-file-no-firmware.txt)
- [debug-file-faulty-firmware.txt](https://github.com/Nitrokey/nitrokey-app/files/2752695/debug-file-faulty-firmware.txt)

### Faulty firmware dump

[firmware.bin.zip](https://github.com/Nitrokey/nitrokey-app/files/2752696/firmware.bin.zip)

szszszsz, 2019-01-18:
Moved to Storage firmware repository.

szszszsz, 2019-01-18:
Retested under Fedora 29 with https://github.com/Nitrokey/libnitrokey/commit/a2b7b9aeb343a10b58ba84ef7ebd0eaff9f0ad27.

In a stress test it turned out, that overall success rate for export is 3/10 and 5/20.
When before exporting was performed the UV unmount in the host OS, the success rate went up to 9/10 and 20/20.
It seems that the auto-remounting side-effect, introduced in v0.46, highly influences this feature. Probably host access to the UV conflicts with the write by the device, thus failing to finish the export.

Solution:
1. Fix by removing the direct cause - correctly remove the UV by making its capacity == 0, or using other ways to inform the OS about the block device removal.
2. Add error codes handling for the firmware exporting procedure, to inform about file creation failure.

szszszsz, 2019-01-18:
Thank you for the feedback!
The firmware verification guide should be updated based on this to provide the workaround, until the issue will be fixed.
cc @alex-nitrokey 

Related: https://github.com/Nitrokey/nitrokey-storage-firmware/issues/45

muellermartin, 2019-01-18:
Based on your stress test I've added a similar test for macOS (see PR Nitrokey/libnitrokey#151).

The results for me:
- with unmount 19/20 succeed
- without unmount 0/20 succeed

szszszsz, 2019-01-31:
@NKelias Could you take a brief look to that, if time would permit?