Nitrokey/nitrokey-storage-firmware #101: Enumeration on 0.55 Firmware (Nitrokey Storage 2)

Unb9rn, 2021-02-01:
After updating my Nitrokey Storage 2 to 0.55 I have some enumeration problems, which are reproduced on different machines and OSes.
After plugging in the token nothing would happen, no logs in system journal, no USB storage device or cardreader getting detected.
After some unplug-replug iterations, it would enumerate. Tried reflashing in Linux and in Windows, tried factory-resetting, formatting unencrypted storage-nothing seems to help.

szszszsz, 2021-02-01:
Hi!
1. What was your previous firmware version?
2. Have you tried downgrading to last working?

Unb9rn, 2021-02-02:
It was on 0.54 Firmware, rolling back to 0.54 seems to solve this problem.