Nitrokey/nitrokey-storage-firmware #113: Firmware binaries mismatch with freshly flashed firmware

Jayman2000, 2021-07-31:
I just flashed [v0.56](https://github.com/Nitrokey/nitrokey-storage-firmware/releases/tag/V0.56), and when I try to verify it, I get `Firmware binaries mismatch`.

### Steps to Reproduce
1. Download the firmware and verify it using `gpg` and `sha256sum`.
```
[jayman@Jason-Desktop-Linux nkcompare]$ gpg --verify sha256sum.sig 
gpg: WARNING: unsafe permissions on homedir '/home/jayman/.gnupg'
gpg: assuming signed data in 'sha256sum'
gpg: Signature made Sat 31 Jul 2021 11:48:36 AM EDT
gpg:                using RSA key 868184069239FF65DE0BCD7DD9BAE35991DE5B22
gpg: Good signature from "Szczepan Zalega <szczepan.zalega@gmail.com>" [unknown]
gpg:                 aka "Szczepan Zalega (Nitrokey) <szczepan@nitrokey.com>" [unknown]
gpg: WARNING: This key is not certified with a trusted signature!
gpg:          There is no indication that the signature belongs to the owner.
Primary key fingerprint: 8681 8406 9239 FF65 DE0B  CD7D D9BA E359 91DE 5B22
[jayman@Jason-Desktop-Linux nkcompare]$ sha256sum -c sha256sum
sha256sum: nitrokey-storage-V0.56-reproducible.hex: No such file or directory
nitrokey-storage-V0.56-reproducible.hex: FAILED open or read
nitrokey-storage-V0.56.hex: OK
sha256sum: WARNING: 1 listed file could not be read
[jayman@Jason-Desktop-Linux nkcompare]$
```
2. [Flash the firmware](https://docs.nitrokey.com/storage/linux/firmware-update.html).
```
[jayman@Jason-Desktop-Linux nkcompare]$ sudo dfu-programmer at32uc3a3256s erase
Checking memory from 0x2000 to 0x3FFFF...  Not blank at 0x2001.
Erasing flash...  Success
Checking memory from 0x2000 to 0x3FFFF...  Empty.
[jayman@Jason-Desktop-Linux nkcompare]$ sudo dfu-programmer at32uc3a3256s flash --suppress-bootloader-mem nitrokey-storage-V0.56.hex 
Checking memory from 0x2000 to 0x3DDFF...  Empty.
0%                            100%  Programming 0x3BE00 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
0%                            100%  Reading 0x3E000 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
Validating...  Success
0x3BE00 bytes written into 0x3E000 bytes memory (96.57%).
[jayman@Jason-Desktop-Linux nkcompare]$ sudo dfu-programmer at32uc3a3256s launch
[jayman@Jason-Desktop-Linux nkcompare]$
```
3. Wait for the unencrypted volume to reappear.
4. If necessary, create a filesystem that the firmware can be written to.
```
[jayman@Jason-Desktop-Linux nkcompare]$ sudo wipefs -a -f /dev/sdc
/dev/sdc: 8 bytes were erased at offset 0x00000052 (vfat): 46 41 54 33 32 20 20 20
/dev/sdc: 1 byte was erased at offset 0x00000000 (vfat): eb
/dev/sdc: 2 bytes were erased at offset 0x000001fe (vfat): 55 aa
[jayman@Jason-Desktop-Linux nkcompare]$ sudo mkfs.vfat -F 32 -n NITRO /dev/sdc
mkfs.fat 4.2 (2021-01-31)
[jayman@Jason-Desktop-Linux nkcompare]$
```
5. Open the Nitrokey App.
6. (Optional) In the overview window, go to Menu>About Nitrokey to check the firmware version.
6. In the overview window, click Menu>Configure>Export firmware to file.
7. Enter the Admin PIN and press OK.
8. Once the firmware has been exported, [verify it using nkcompare](https://github.com/Nitrokey/nitrokey-storage-firmware/tree/master/tools).
```
[jayman@Jason-Desktop-Linux nkcompare]$ ./nkcompare.sh nitrokey-storage-V0.56.hex firmware.bin 
Running ./nkcompare.sh nitrokey-storage-V0.56.hex firmware.bin
Firmware binaries mismatch
Original Firmware SHA256: 5a8c33d989c2756461322d4b3479555214f2a612fb619fbee11e420e44c0000c
Exported Firmware SHA256: 28372752dcd4ed1f70253e35e671adbd5c0f0c7bea4754381f77ff2ea00bbf3b
[jayman@Jason-Desktop-Linux nkcompare]$
```

szszszsz, 2021-08-09:
Hi!
Confirmed, with the exact hash sums on my side. Looking into it.

szszszsz, 2021-08-09:
Looks like the last ~3.5kB of the firmware is not exported due to rounding error in the write function (to the full 4kB blocks).
In this release maximum firmware size was changed to 495 pages, which is:
```
(0x00040000 - (512-495)*512) / 1024
247.5                              
```

And the exported length is 244 kB.
See src/USER_INTERFACE/file_io.c:290:

```
    // Write data
    for (i = 0; i < (FILEIO_IMAGE_SIZE / FILEIO_IMAGE_BLOCKSIZE); i++)
    {
        CI_LocalPrintf ("Write block = %3d\r\n", i);
        Ret_s32 = write (FileID_s32, (u8 *) (FILEIO_IMAGE_START + i * FILEIO_IMAGE_BLOCKSIZE), FILEIO_IMAGE_BLOCKSIZE);
    }
```

Solutions:
1. Move limit 1 page up to 248kB (if possible), or
2. Handle remainder in the writing function

szszszsz, 2021-08-10:
Corrected. Will be working again from the next release.