Nitrokey/nitrokey-storage-firmware #118: V0.5 Security Bug - Affected Devices List

r0kettherac00n, 2022-10-18:
I recall that there was a security issue relating to empty AES key being used to encrypt data if a reset was done via GnuPG.
Any idea which batch of devices were shipped with firmware above above 0.50?

szszszsz, 2022-10-18:
1. The firmware is upgradeable, hence it should not be a problem.
2. Expectation is, that most of the data will be physically overwritten. Whatever the microcontroller use for the spare space, this should not be left untouched. Perhaps running full procedure two times might be even better.

r0kettherac00n, 2022-10-18:
Thank man, was just wondering if the nitrokeys shipped from 2020 onwards after the bug was discovered came with firmware above 0.50

szszszsz, 2022-10-18:
Yes. We are shipping with the latest right now, and updated firmware in production asap. I do not have exact stats at hand, but we encourage users to run the firmware updates, especially when there is a serious problem like this one.

r0kettherac00n, 2022-10-18:
Nice, might be better to maybe add a statement ike
"Shipments from xxxx-xx-xx onwards were unaffected by this issue"

closed 2022-10-18 by r0kettherac00n