Nitrokey/nitrokey-storage-firmware #121: 'avr32_pm_t' has no member named 'bod33'

Pekkari, 2023-08-08:
Hi,

Following the instructions on the readme, I setup a build environment so that I can make a firmware build of v0.57
with a small change at conf_sd_mmc_mci.h to increase the size of the unencrypted partition. After succesfully
setting up the avr tools(studio 2.6 and toolchain 2.4.2) in a rhel 5 environment, I reach the comand line instruction:
```cd src && make CC=path/to/avr32-gcc```

and the following output comes out:
```
nitrokey.c:125: warning: ignoring #pragma mark 
nitrokey.c:126: warning: initialization makes pointer from integer without a cast
nitrokey.c:128: warning: ignoring #pragma mark 
nitrokey.c: In function 'pm_bod33_get_level':
nitrokey.c:572: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c:572: error: 'AVR32_PM_BOD33_LEVEL_MASK' undeclared (first use in this function)
nitrokey.c:572: error: (Each undeclared identifier is reported only once
nitrokey.c:572: error: for each function it appears in.)
nitrokey.c:572: error: 'AVR32_PM_BOD33_LEVEL_OFFSET' undeclared (first use in this function)
nitrokey.c:573: warning: control reaches end of non-void function
nitrokey.c: In function 'pm_bod33_set_level':
nitrokey.c:590: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c:592: error: 'AVR32_PM_BOD33_LEVEL_MASK' undeclared (first use in this function)
nitrokey.c:599: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c:604: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c: In function 'pm_bod33_ResetEnable':
nitrokey.c:620: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c:621: error: 'AVR32_PM_BOD33_CTRL_MASK' undeclared (first use in this function)
nitrokey.c:622: error: 'AVR32_PM_BOD33_CTRL_OFFSET' undeclared (first use in this function)
nitrokey.c:628: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c:633: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c:635: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c: In function 'pm_bod33_ResetDisable':
nitrokey.c:651: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c:652: error: 'AVR32_PM_BOD33_CTRL_MASK' undeclared (first use in this function)
nitrokey.c:653: error: 'AVR32_PM_BOD33_CTRL_OFFSET' undeclared (first use in this function)
nitrokey.c:659: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c:664: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c: In function 'main':
nitrokey.c:802: warning: unused variable 'task_woken'
make: *** [nitrokey.o] Error 1
```
Is there anything I can be missing in my environment so that this definition is not found
appropriately, or is this a legit bug?

Thanks!

szszszsz, 2023-08-09:
Hi!

1. Did you copy the `pm_240.h` file (see https://github.com/Nitrokey/nitrokey-storage-firmware#gui, step 5) ? It looks like it is missing from the error message.
2. As a last resort, the Nix team has packaged firmware building for Nitrokey Storage under the link below. Perhaps
    this will help resolve problems, or allow for easier build environment setup. The mentioned link:
    - https://github.com/ngi-nix/nitrokey-firmware/blob/master/devices/nitrokey-storage.nix
    - https://github.com/ngi-nix/nitrokey-firmware/tree/master (build instructions)

Pekkari, 2023-08-09:
Just in case I re-copied the file to my ide, and run the make step again, the result is still
a bod33 missing. I indeed find the definition [here](https://github.com/Nitrokey/nitrokey-storage-firmware/blob/d59251d3d0839323009d4c369c17fb7b8423c577/pm_240.h#L1345), changing the avr32_pm_bod33_t definition
from capital to lower case makes no difference.

szszszsz, 2023-08-09:
Can you double check, that you have replaced the `pm_231.h` with this `*240.h`?

szszszsz, 2023-08-09:
I have just checked and updated this Dockerfile and all builds for me. I have not flashed the build binary though.
- https://github.com/Nitrokey/nitrokey-storage-firmware/blob/master/Dockerfile

Pekkari, 2023-08-09:
Here is some output of a session I just did to let you revise I'm doing it good:
```
$ pwd
/home/rhel/nitrokey/nitrokey-storage-firmware
$ env | grep PATH
T_SUBPATH=plugins/com.atmel.avr.toolchains.linux.x86_64_3.0.0.201009140852/os/linux/x86_64/avr32/include/avr32
TOOLCHAIN_PATH=/home/rhel/nitrokey/as4e-ide
PATH=/usr/kerberos/bin:/usr/local/bin:/usr/bin:/bin:/usr/X11R6/bin:/home/rhel/bin
$ git diff pm_240.h
$ diff pm_240.h /home/rhel/nitrokey/as4e-ide/plugins/com.atmel.avr.toolchains.linux.x86_64_3.0.0.201009140852/os/linux/x86_64/avr32/include/avr32/pm_231.h
$ cd src
$ make CC=/usr/bin/avr32-gcc
*** nitrokey.c > nitrokey.o
nitrokey.c:125: warning: ignoring #pragma mark 
nitrokey.c:126: warning: initialization makes pointer from integer without a cast
nitrokey.c:128: warning: ignoring #pragma mark 
nitrokey.c: In function 'pm_bod33_get_level':
nitrokey.c:572: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c:572: error: 'AVR32_PM_BOD33_LEVEL_MASK' undeclared (first use in this function)
nitrokey.c:572: error: (Each undeclared identifier is reported only once
nitrokey.c:572: error: for each function it appears in.)
nitrokey.c:572: error: 'AVR32_PM_BOD33_LEVEL_OFFSET' undeclared (first use in this function)
nitrokey.c:573: warning: control reaches end of non-void function
nitrokey.c: In function 'pm_bod33_set_level':
nitrokey.c:590: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c:592: error: 'AVR32_PM_BOD33_LEVEL_MASK' undeclared (first use in this function)
nitrokey.c:599: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c:604: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c: In function 'pm_bod33_ResetEnable':
nitrokey.c:620: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c:621: error: 'AVR32_PM_BOD33_CTRL_MASK' undeclared (first use in this function)
nitrokey.c:622: error: 'AVR32_PM_BOD33_CTRL_OFFSET' undeclared (first use in this function)
nitrokey.c:628: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c:633: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c:635: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c: In function 'pm_bod33_ResetDisable':
nitrokey.c:651: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c:652: error: 'AVR32_PM_BOD33_CTRL_MASK' undeclared (first use in this function)
nitrokey.c:653: error: 'AVR32_PM_BOD33_CTRL_OFFSET' undeclared (first use in this function)
nitrokey.c:659: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c:664: error: 'avr32_pm_t' has no member named 'bod33'
nitrokey.c: In function 'main':
nitrokey.c:802: warning: unused variable 'task_woken'
make: *** [nitrokey.o] Error 1
$ yum list installed | grep avr
avr32-binutils.i386                       2.19.atmel.1.2.0-1.el5       installed
avr32-buildroot-essentials.i386           1.0.0-1.el5                  installed
avr32-gcc-newlib.i386                     4.3.2-atmel.1.2.0_1.el5      installed
avr32-gdb.i386                            6.7.1.atmel.1.0.4-1.el5      installed
avr32-gnu-toolchain.i386                  2.4.2-1.el5                  installed
avr32gdbproxy.i386                        4.1.0-1.el5                  installed
avr32headers.noarch                       2.3.4-1.el5                  installed
avr32parts.noarch                         2.4.0-1.el5                  installed
avr32program.i386                         4.1.0-1.el5                  installed
avr32trace.i386                           2.2.0-1.el5                  installed
avrfwupgrade.i386                         2.0.0-1.el5                  installed
libavr32ocd.i386                          4.1.0-1.el5                  installed
libavr32sim.i386                          0.3.0-1.el5                  installed
libavrtools.i386                          4.1.0-1.el5                  installed
$ uname -a
Linux localhost.localdomain 2.6.18-398.el5 #1 SMP Tue Aug 12 06:26:17 EDT 2014 x86_64 x86_64 x86_64 GNU/Linux
```

szszszsz, 2023-08-09:
Great to hear that! 
I have no clue, why it does not work for you in the previous attempt. One last idea is, that `make CC=/usr/bin/avr32-gcc` is using wrong compiler path, which does not have the header file in question updated or available.
Either way closing as working. Happy hacking!

szszszsz, 2023-08-09:
Final note: newer avr32 compilers do not have this MCU supported AFAIK. Please let me know if you would found otherwise.